mit-han-lab/bevfusion #2: About the inference time

zyl1336110861, 2022-06-03:
@zhijian-liu Thank you very much for your sharing! I am reading your paper recentely. I see that your optimized model can run about 100ms per frame. So I am curious about what is the efficiency bottleneck now. Is it the Swin-T process?

kentang-mit, 2022-06-03:
Hello @zyl1336110861, thanks for your interest in our work! The efficiency bottleneck is the LiDAR backbone, which is a SparseConvNet. Previously we also study the optimization of SparseConv in the system-level and hardware-level. You are welcomed to check out our recent papers [TorchSparse](https://torchsparse.mit.edu) and [PointAcc](https://pointacc.mit.edu).

zyl1336110861, 2022-06-03:
@kentang-mit Ok! I will refer to these papers! Thank you very much!

kentang-mit, 2022-06-03:
You are welcome.

closed 2022-06-03 by kentang-mit

zyl1336110861, 2022-06-06:
@kentang-mit Hi, please see here [,](https://github.com/traveller59/spconv/issues/383) is that true? It says that [spconv2.x](https://github.com/traveller59/spconv) is greatly faster than torchsparse.

zyl1336110861, 2022-06-06:
@kentang-mit Hi, please see [here](https://github.com/traveller59/spconv/issues/383 ) , is that true? It says that [spconv2.x](https://github.com/traveller59/spconv) is greatly faster than torchsparse.