mit-han-lab/bevfusion #3: Which version of mmdetection3d is bevfusion based on?

mljack, 2022-06-04:
Could you provide the version or commit it forks from?

Thanks a lot!

kentang-mit, 2022-06-04:
Hi @mljack, if I remember it correctly, we forked our repo from [this commit](https://github.com/open-mmlab/mmdetection3d/tree/60ce864ff76af4316fb9ae56a2a5b7741bfdd9ab).

mljack, 2022-06-04:
Is it possible to public more training details for these pretrained weights? For example, the training machine GPU count, training time or even training log?

kentang-mit, 2022-06-04:
Definitely. We train BEVFusion on 8 NVIDIA RTX A6000 GPUs, each with 48G GPU memory. The detection model takes around 35 hours to train and the segmentation model takes 25 hours to train. The training log will be included in future release. We are still working hard to make BEVFusion trainable on GPUs with less memory. Stay tuned for more updates!

mljack, 2022-06-05:
Super cool! Looking forward your new updates!

closed 2022-06-05 by mljack