mit-han-lab/bevfusion #6: python3 setup.py develop failed with  cuda13 + pytorch1.11

xczhanjun, 2022-06-06:
#ENV 
  conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch
  pip3 install torch torchvision torchaudio --extra-index-url https://download.pytorch.org/whl/cu113

#ERROR
mmdet3d/ops/group_points/src/group_points.cpp:4:10: fatal error: THC/THC.h: No such file or directory
    4 | #include <THC/THC.h>
      |          ^~~~~~~~~~~
compilation terminated.

xczhanjun, 2022-06-06:
@zhijian-liu could you setup an docker env ?

kentang-mit, 2022-06-06:
Hi @xczhanjun,

Would you mind trying out PyTorch 1.9 or PyTorch 1.10? This header might be removed in 1.11. If that's the problem I will update the backend.

Best,
Haotian

xczhanjun, 2022-06-06:
I tried PyTorch 1.10 with cuda113,  it works.  I will send you the requiremensts.txt as pull request. @kentang-mit