mit-han-lab/bevfusion #10: Problem while evaluation

Etah0409, 2022-06-07:
Hi, your work is really nice. When I try to use this command:
`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`

RuntimeError: Address already in use

The traceback is:
`Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 115, in main
    dist.init()
  File "/data/workspace/bevfusion/torchpack-master/torchpack/distributed/context.py", line 35, in init
    torch.distributed.init_process_group(
  File "/data/workspace/.conda/envs/bev/lib/python3.8/site-packages/torch/distributed/distributed_c10d.py", line 520, in init_process_group
    store, rank, world_size = next(rendezvous_iterator)
  File "/data/workspace/.conda/envs/bev/lib/python3.8/site-packages/torch/distributed/rendezvous.py", line 142, in _tcp_rendezvous_handler
    store = TCPStore(result.hostname, result.port, world_size, start_daemon, timeout)
RuntimeError: Address already in use`

I'm not familiar with mpi, so I didn't find the code that can change the port. The openmpic version is 4.1.4, and the torchpack version is the latest version (install from source code).

A lot of thanks for your kindly reply :)

kentang-mit, 2022-06-07:
Hi @Etah0409,

Thanks for your interest. May I ask whether there is other distributed jobs running on your machine when you are launching this command? Also, I'm inviting @zhijian-liu to this issue, who is more familiar with the torchpack framework.

Best,
Haotian

kentang-mit, 2022-06-07:
In general we don't recommend launching two distributed jobs simultaneously. Maybe Zhijian can provide better suggestions on specifying a new port.

songlilucky, 2022-06-08:
I find torchpack.distributed initprocess group with TCP method, So you must set env `export MASTER_HOST={your host ip:port}`, such as `tcp://10.2.22.122:22111`

Etah0409, 2022-06-08:
@songlilucky Thanks! That solves my problem.

closed 2022-06-08 by Etah0409

qfwysw, 2022-08-10:
> export MASTER_HOST={your host ip:port}

Hi, I have the same promble, but I don't know how to modify it. Can you give me more details?

wanglaotou, 2022-09-07:
i met 

> > export MASTER_HOST={your host ip:port}
> 
> Hi, I have the same promble, but I don't know how to modify it. Can you give me more details?

you can use single gpu test, i use this way worked!