mit-han-lab/bevfusion #9: How to build the bev pooling in my own project?

exiawsh, 2022-06-07:
Hello, thanks for your great work!
If I want to only build the bev pooling operation in my own project, how could I do it?

kentang-mit, 2022-06-07:
Hello @exiawsh,

Thanks for your interest. If you are using mmdetection3d, I believe this modification in the [setup.py](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/setup.py#L76) file is helpful.

Related to the BEV pooling operation itself, I recommend you have a look at [this function](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/mmdet3d/models/vtransforms/base.py#L137), in which line 167 is the call to the CUDA kernel. It has 6 parameters, namely:

- x: input features, of size [N_camera_feats, C] (filtered from the [B, N_cam, C, H_cam, W_cam] feature map)
- geom_feats: 3D coordinates of each input feature, of size [N_camera_feats, 4]
- B: batch size
- Z: the z-size of the BEV feature map
- H_BEV: the x-size of the BEV feature map
- W_BEV: the y-size of the BEV feature map.

The kernel will return a feature tensor of size [B x C x Z x H_BEV x W_BEV]. What the CUDA kernel does is to sum up the features with the same 3D coordinates and insert them to the correct memory location. 

Best,
Haotian

exiawsh, 2022-06-07:
> 

Thank you for your patience and great work! In fact, I have no previous experience in CUDA Programming. I will try to compare the implementation of mmdetection. In addition, may you recommend some CUDA Programming learning materials?

exiawsh, 2022-06-07:
> Hello @exiawsh,
>
> Thanks for your interest. If you are using mmdetection3d, I believe this modification in the [...]

Thank you for your patience and great work! In fact, I have no previous experience in CUDA Programming. I will try to compare the implementation of mmdetection. In addition, may you recommend some CUDA Programming learning materials?

kentang-mit, 2022-06-07:
Sure, you can have a look at [this repo](https://github.com/charlesq34/pointnet2) and [this repo](https://github.com/erikwijmans/Pointnet2_PyTorch) for some important kernels used in point cloud processing.

closed 2022-06-09 by kentang-mit