mit-han-lab/bevfusion #11: Problem with create data

songlilucky, 2022-06-08:
```
File "bevfusion/tools/data_converter/create_gt_database.py", line 275, in create_groundtruth_database
    image_idx = example["sample_idx"]
KeyError: 'sample_idx'
```
 I  find that code Collect3D class in  `/mmdet3d/datasets/pipelines/formating.py` is different from origin code in mmdetection3d . The new code will miss the key `sample_idx`.

A lot of thanks for your kindly reply :)

xczhanjun, 2022-06-08:
@songlilucky yes, I found the same problem . 
Please add and edit two lines in 
mmdet3d/datasets/nuscenes_dataset.py
def get_data_info(self, index: int) -> Dict[str, Any]:
        info = self.data_infos[index]
        data = dict(
            token=info["token"],
 +++++++++   sample_idx=info['token'],

and 
change this line in the same function:
camera_intrinsics[:3, :3] = camera_info["cam_intrinsics"]
=====> 
camera_intrinsics[:3, :3] = camera_info["camera_intrinsics"]

songlilucky, 2022-06-08:
> camera_intrinsics

 Thanks! That solves the problem!

closed 2022-06-08 by songlilucky

sborse3, 2022-06-08:
@zhijian-liu, @xczhanjun thanks for the great work and this fix. I faced this issue as well, and it fixed my problem