mit-han-lab/bevfusion #12: Problem when evaluation

Capchenxi, 2022-06-09:
Hello,

I'm working with python3.8+torch1.10.0+cuda111+ cuda driver version460.32.03+TeslaV100

It works well when creating data, but it will give  CUDA_ERROR: no kernel image is available for execution. (This error happens after a .contiguous() operation in bev pooling func).   I searched that problem and got that it may be caused by the version conflict of  cuda/torch/mmcv.  So I'm wondering is there any limitation of the envs? What is the version of cuda are you using? Thanks!

kentang-mit, 2022-06-09:
Hi @Capchenxi,

Perhaps you can try modifying [this line](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/setup.py#L22). I remember the arch. of V100 is sm_70 and compute_70 (please double check). Just add one more line will help.

Best,
Haotian

Capchenxi, 2022-06-09:
> Hi @Capchenxi,
> 
> Perhaps you can try modifying [this line](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/setup.py#L22). I remember the arch. of V100 is sm_70 and compute_70 (please double check). Just add one more line and rerun
> 
> ```shell
> python setup.py develop
> ```
> 
> will help.
> 
> Best, Haotian

Thanks for your prompt reply and adding the arch in setup.py solved my problem. 
Many thanks!

Best,
Chenxi

closed 2022-06-09 by Capchenxi