mit-han-lab/bevfusion #13: Meet a problem when running Nuscenes data preparation script.

PigLogic-Cyber, 2022-06-09:
Environment:
```
Python = 3.8
PyTorch = 1.10.1
tqdm = 4.64.0
torchpack = 0.3.1
mmcv-full = 1.4.0
mmdet = 2.20.0
mmdet3d = 0.0.0
torchvision = 0.11.2
```
Description:
After I prepare the develop environment and download the Nuscenes original dataset, I try to run the command:`python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes`, but I meet an **error** :`Traceback (most recent call last):
  File "tools/create_data.py", line 3, in <module>
    from tools.data_converter import nuscenes_converter as nuscenes_converter
  File "/home/user/BEVFusion/tools/data_converter/nuscenes_converter.py", line 13,                                                                                                                         in <module>
    from mmdet3d.core.bbox.box_np_ops import points_cam2img
  File "/home/user/BEVFusion/mmdet3d/core/__init__.py", line 2, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/home/user/BEVFusion/mmdet3d/core/bbox/__init__.py", line 3, in <module>
    from .iou_calculators import (AxisAlignedBboxOverlaps3D, BboxOverlaps3D,
  File "/home/user/BEVFusion/mmdet3d/core/bbox/iou_calculators/__init__.py", line                                                                                                                         1, in <module>
    from .iou3d_calculator import (
  File "/home/user/BEVFusion/mmdet3d/core/bbox/iou_calculators/iou3d_calculator.py                                                                                                                        ", line 5, in <module>
    from ..structures import get_box_type
  File "/home/user/BEVFusion/mmdet3d/core/bbox/structures/__init__.py", line 1, in                                                                                                                         <module>
    from .base_box3d import BaseInstance3DBoxes
  File "/home/user/BEVFusion/mmdet3d/core/bbox/structures/base_box3d.py", line 5,                                                                                                                         in <module>
    from mmdet3d.ops.iou3d import iou3d_cuda
  File "/home/user/BEVFusion/mmdet3d/ops/__init__.py", line 34, in <module>
    from .roiaware_pool3d import (
  File "/home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/__init__.py", line 1, in                                                                                                                         <module>
    from .points_in_boxes import points_in_boxes_batch, points_in_boxes_cpu, points_in_boxes_gpu
  File "/home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/points_in_boxes.py", line                                                                                                                         3, in <module>
    from . import roiaware_pool3d_ext
ImportError: /home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.                                                                                                                        cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c1015SmallVectorBaseIjE8grow_podEPvmm
`. 
Many thanks for your kindly help!

PigLogic-Cyber, 2022-06-09:
BTW, `cudatoolkit=11.3.1`

PigLogic-Cyber, 2022-06-10:
> ```python
> from mmdet3d.ops.iou3d import iou3d_cuda
> ```

Dear Haotian,
    First of all, thanks for your kindly reply. 
    1. I tried `from mmdet3d.ops.iou3d import iou3d_cuda` in a Python command line interface, it reply as `>>> from mmdet3d.ops.iou3d import iou3d_cuda
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/user/BEVFusion/mmdet3d/ops/__init__.py", line 34, in <module>
    from .roiaware_pool3d import (
  File "/home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/__init__.py", line 1, in <module>
    from .points_in_boxes import points_in_boxes_batch, points_in_boxes_cpu, points_in_boxes_gpu
  File "/home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/points_in_boxes.py", line 3, in <module>
    from . import roiaware_pool3d_ext
ImportError: /home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c1015SmallVectorBaseIjE8grow_podEPvmm
`， same error.
    2. The server is composed of two Nvidia GeForce 3090

kentang-mit, 2022-06-10:
Hi @PigLogic-Cyber,

It looks like this `roiaware_pool3d` kernel is not compiled successfully. I did try the same build command on a server with multiple 3090 GPUs, and it seems that I can successfully import this backend function. I would suggest you to delete `/home/user/BEVFusion/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.cpython-38-x86_64-linux-gnu.so`, and rerun  ```bash
python setup.py develop
```
under the root folder to see whether it can resolve the issue. 

Best,
Haotian

kentang-mit, 2022-06-10:
By the way, if this still cannot solve your problem, would you mind sending us the command line output when you run 
```bash
python setup.py develop
```

This might be helpful for us to diagnose the problem.

Best,
Haotian

PigLogic-Cyber, 2022-06-13:
> By the way, if this still cannot solve your problem, would you mind sending us the command line output when you run
> 
> ```shell
> python setup.py develop
> ```
> 
> This might be helpful for us to diagnose the problem.
> 
> Best, Haotian

Dear KenTang,
   First of all, thanks for your patient and reply. I do find some **warning** saying that the cudatoolkit version is mismatch(original version is 11.3), so I downgrade the version of cudatoolkit, right now my environment is like `Python = 3.8.13
PyTorch = 1.10.1
tqdm = 4.64.0
torchpack = 0.3.1
mmcv-full = 1.4.0
mmdet = 2.20.0
mmdet3d = 0.0.0
torchvision = 0.11.2
cudatoolkit = 11.1.1`
Then I run the command `python setup.py develop` the output is `running develop
/home/user/anaconda3/envs/fusion/lib/python3.8/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
warnings.warn(
/home/user/anaconda3/envs/fusion/lib/python3.8/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
warnings.warn(
running egg_info
writing mmdet3d.egg-info/PKG-INFO
writing dependency_links to mmdet3d.egg-info/dependency_links.txt
writing top-level names to mmdet3d.egg-info/top_level.txt
/home/user/anaconda3/envs/fusion/lib/python3.8/site-packages/torch/utils/cpp_extension.py:381: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'mmdet3d.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'mmdet3d.egg-info/SOURCES.txt'
running build_ext
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/spconv/sparse_conv_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/spconv
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/bev_pool/bev_pool_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/bev_pool
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/iou3d/iou3d_cuda.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/iou3d
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/voxel/voxel_layer.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/voxel
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/roiaware_pool3d/roiaware_pool3d_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/roiaware_pool3d
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/ball_query/ball_query_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/ball_query
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/knn/knn_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/knn
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/paconv/assign_score_withk_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/paconv
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/group_points/group_points_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/group_points
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/interpolate/interpolate_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/interpolate
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/furthest_point_sample/furthest_point_sample_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/furthest_point_sample
copying build/lib.linux-x86_64-3.8/mmdet3d/ops/gather_points/gather_points_ext.cpython-38-x86_64-linux-gnu.so -> mmdet3d/ops/gather_points
Creating /home/wangruzhuo/anaconda3/envs/fusion/lib/python3.8/site-packages/mmdet3d.egg-link (link to .)
mmdet3d 0.0.0 is already the active version in easy-install.pth

Installed /home/user/BEVFusion
Processing dependencies for mmdet3d==0.0.0
Finished processing dependencies for mmdet3d==0.0.0
`

kentang-mit, 2022-06-13:
Hi @PigLogic-Cyber,

Have you tried deleting the `build` folder and recompile? It seems to me that the setup.py script did nothing but copied pre-built results around. However, as you have mentioned, you changed the version of the CUDA toolkit. In this case, I believe a re-compilation is necessary.

Related to the library version, actually I'm also using PyTorch 1.10.1 + CUDA devkit 11.3. So I don't think your library version is problematic.

If thing still cannot work, I would suggest you to try compiling other mmdetection3d-based repos, such as [DETR3D](https://github.com/WangYueFt/detr3d). I'm curious whether similar error would occur again when you

```python
from mmdet3d.ops.iou3d import iou3d_cuda
```

Best,
Haotian

PigLogic-Cyber, 2022-06-13:
> Hi @PigLogic-Cyber,
>
> Have you tried deleting the `build` folder and recompile? It seems to me that the setup.py script did nothing but copied pre-built results around. However, as you have [...]

Dear Haotian,
After deleting the `build ` dir and recompile the development environment, problem solved. So my original problem was caused by the higher version of pytorch(1.11.0). Then the problem happended since I downgraded the pytorch version but forgot to **delete** `build ` dir before compiling the development environment. Really save my day, thx a lot!
Best,
Roy

closed 2022-06-13 by PigLogic-Cyber