mit-han-lab/bevfusion #14: How to test the model inference time?

zyl1336110861, 2022-06-09:
@songhan Hi, When I want to use the _**benchmark.py**_ script to test the inference speed, I encountered the following error:

[image]

or 

[image]

Can you provide a completed script to test inference time? Thank you very much!

kentang-mit, 2022-06-09:
Hi @zyl1336110861,

We did not use the `benchmark` script to test the inference time. Instead, we manually insert timers into [this file](mmdet3d/models/fusion_models/bevfusion.py) to calculate the runtime of different parts of our model and sum the results up to get the final latency. In this process, we ignore the voxelization time because we assume it to be a pipeline-able procedure and is very fast after the dynamic voxelization optimization proposed in MVF. The way to add a timer to one module is:

```python
torch.cuda.synchronize()
st = time.time()

# run the module
x = module(x)

torch.cuda.synchronize()
ed = time.time()
```

And `ed-st` will be the time you want. For results reported in the paper, they are averaged across the entire validation set, because the runtime of SparseConv modules can vary for different samples.

Best,
Haotian

zyl1336110861, 2022-06-10:
@kentang-mit Thanks for your patience. I just test the inference use the v1.0-mini dataset at a single RTX3090. The inference time is 230ms which is much slower than your paper. Maybe the  test data is small or the voxelization is slow.

kentang-mit, 2022-06-10:
Yes. The current voxelization implementation can take ~100ms. As I have mentioned, this procedure can be replaced with dynamic voxelization, and it can be pipelined (so there is no cost).

zyl1336110861, 2022-06-10:
OK! Thank you very much!

kentang-mit, 2022-06-10:
You are welcome! I'll close this issue temporarily, feel free to reopen if you have further questions.

closed 2022-06-10 by kentang-mit

JoeyforJoy, 2022-06-18:
This error may due to the wrong use of mmcv.Config. 
I replace the Line27 as below and import necessary modules. Then I can run `benchmark.py` successfully.
```python
...

from torchpack.utils.config import configs
from mmdet3d.utils import recursive_eval

...

configs.load(args.config, recursive=True)
cfg = Config(recursive_eval(configs), filename=args.config)

# set cudnn_benchmark
...
```