mit-han-lab/bevfusion #19: Question about the improvement calculation of multi-modality in ablation studies?

yt7589, 2022-06-11:
In the paper 2205.13542 ablation studies section in the table 4.a, it said: the NDS score is 40.2 when use Camera mode only, and the NDS score is 69.5 when use Camera + Lidar mode. According to my understanding the improvement rate should be:
$$r =\frac{69.5 - 40.2}{40.2} = 0.73$$
But in the paper you said the improvement ratio is 6.6%. My question is How you get this number (6.6%) given the data in table 4.a?

kentang-mit, 2022-06-11:
Hi @yt7589,

I think the 6.6% improvement refers to the BEVFusion (C+L) v.s. the camera-only baseline in the segmentation setting. In this paper we assume that if you want to compare fusion models with camera-only methods, you should use segmentation as the task. This is because camera-only detection mAP/NDS is a bit too low.

Best,
Haotian

kentang-mit, 2022-06-12:
I'll temporarily close this issue. Feel free to reopen if you have further questions.

closed 2022-06-12 by kentang-mit