mit-han-lab/bevfusion #25: HI,train error

zouzouwei, 2022-06-17:
when  run the code:
  torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml 

An error has occurred:
[image]

plz,tell me solve this problem

kentang-mit, 2022-06-17:
Hi @zouzouwei,

I would recommend you to take a look at the discussions in [this issue](https://github.com/mit-han-lab/bevfusion/issues/21) for mpi-related problems. 

Besides, training hyper-parameters / schedules will be released in the future, and we will start to answer questions related to training after that.

Best,
Haotian

zhijian-liu, 2022-06-17:
You need to install `OpenMPI` instead of `MPICH`.

kentang-mit, 2022-06-17:
I'll temporarily close this issue. Please route MPI-related discussions to the aforementioned issue.

closed 2022-06-17 by kentang-mit

hasaikeyQAQ, 2023-04-14:
> You need to install `OpenMPI` instead of `MPICH` to run `torchpack`.

Dear Liu Zhijian,

Thank you for your excellent work. I have a question regarding the usage of distributed training. When I use torchpack -np 1, the training code runs normally, but when I try to use torchpack -np 2, the program stops running without any error messages. My mpi4py version is 3.0.3 and I don't have the openmpi package installed. I'm wondering if this issue with using torchpack distributed training could be caused by not having the openmpi package installed.

Please let me know your thoughts and advice.

Best regards.