mit-han-lab/bevfusion #26: Problem when evaluation

linyanAI, 2022-06-17:


kentang-mit, 2022-06-17:
Hi @linyanAI,

Would you mind helping us locate the exact line of code that causes the error? Is that in the SparseConv backbone or in the view transformer?

Best,
Haotian

linyanAI, 2022-06-17:
File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_func
    output = old_func(*new_args, **new_kwargs)  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/models/vtransforms/base.py", line 167, in bev_pool
    x = bev_pool(x, geom_feats, B, self.nx[2], self.nx[0], self.nx[1])
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/ops/bev_pool/bev_pool.py", line 96, in bev_pool
    x = x.permute(0, 4, 1, 2, 3).contiguous()
RuntimeError: CUDA error: no kernel image is available for execution on the device

It seems in the view transformer, and I check the version in this codebase twice but still got this problem

kentang-mit, 2022-06-17:
Hi @linyanAI,

Thanks for the information. Just to confirm, everything before the BEV pooling kernel (especially the sparse convolution backbone) runs correctly on your machine, right? You can insert a piece of code [here](https://github.com/mit-han-lab/bevfusion/blob/24502fe0979b345ef48e60fbaa45d4d431e9c22d/mmdet3d/models/backbones/sparse_encoder.py#L131):

```python
print(spatial_features)
```

Best,
Haotian

linyanAI, 2022-06-17:
Hi, It didn't return any things. After I check the order of the code, maybe it execute the camera backbone, and then Lidar backbone, but the error happens in the camera stage, so may not return things in lidar stage.

kentang-mit, 2022-06-17:
Hi @linyanAI,

Oh that makes sense to me. I think it might be possible to switch the order of execution [here](https://github.com/mit-han-lab/bevfusion/blob/24502fe0979b345ef48e60fbaa45d4d431e9c22d/mmdet3d/models/fusion_models/bevfusion.py#L167) first like:
```python
for sensor in ["lidar", "camera"]:
```

Let's first make sure whether it is the BEV pooling kernel that cannot run on your device or it is all customized CUDA kernels cannot run on your device.

Best,
Haotian

linyanAI, 2022-06-17:
`  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call
_impl
    return forward_call(*input, **kwargs)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in 
forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call
_impl
    return forward_call(*input, **kwargs)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_fun
c
    output = old_func(*new_args, **new_kwargs)
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 182, in forward
    feature = self.extract_lidar_features(points)
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 123, in extract_lidar_features
    feats, coords, sizes = self.voxelize(x)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 28, in decora
te_context
    return func(*args, **kwargs)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 214, in new_fun
c
    output = old_func(*new_args, **new_kwargs)
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in voxelize
    f, c, n = self.encoders["lidar"]["voxelize"](res)
  File "/mnt/cache/huanglinyan/miniconda3/envs/fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call
_impl
    return forward_call(*input, **kwargs)
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/ops/voxel/voxelize.py", line 131, in forward
    return voxelization(
  File "/mnt/cache/huanglinyan/project/bevfusion/mmdet3d/ops/voxel/voxelize.py", line 55, in forward
    voxel_num = hard_voxelize(
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.`

The same problem encounters in the lidar stage which before the sparse encoder exactly in the voxelize stage.

kentang-mit, 2022-06-17:
Yes, it looks like none of the customized kernels run successfully on your device. I'm also thinking about removing `-gencode=arch=compute_80,code=sm_80` in the `setup.py` in your case. Would you mind giving it a try?

linyanAI, 2022-06-17:
Yes, i can try it, but maybe 7 hours later. And my pytorch install command is as follows:
```
pip install torch==1.10.0+cu111 torchvision==0.11.0+cu111 torchaudio==0.10.0 -f https://download.pytorch.org/whl/torch_stable.html
```

kentang-mit, 2022-06-17:
Got it. Let me know if that does not work.

linyanAI, 2022-06-18:
Thanks for your prompt response! I have solved it by rerun `python setup.py develop` just as you said before.

closed 2022-06-18 by linyanAI

bio-punk, 2023-10-29:
same problem, but I use aarch64 cpu with A100
slove by:
rm -rf build
vi setup.py
change line22 to
           "-gencode=arch=compute_80,code=sm_80",