mit-han-lab/bevfusion #28: Test error: N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0

zhouth-shiny, 2022-06-18:
Thanks for your excellent work！
When I try to evaluate the detection variant of BEVFusion, I meet a problem as follows:

Traceback (most recent call last):
  File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmdet/apis/test.py", line 98, in multi_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 184, in forward
    feature = self.extract_lidar_features(points)
  File "/home/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 125, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/home/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 122, in forward
    x = encoder_layer(x)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/ops/spconv/modules.py", line 135, in forward
    input = module(input)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/ops/spconv/modules.py", line 135, in forward
    input = module(input)
  File "/home/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/home/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: /home/zth/bevfusion/mmdet3d/ops/spconv/include/tensorview/helper_launch.h 17
**N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0**
  
I'm not sure about this problem, hope to get your advice! Thanks! :)

kentang-mit, 2022-06-18:
Hi @zhouth-shiny,

To me, it looks like a problem related to the spconv backend. Would you mind printing out the shape of `feats` and `coords` before the forward function? It will also be helpful if you can help us locate the exact id of the block which throws this error (e.g. is it the first SparseConv layer, or some layers can actually run successfully and some cannot).

Best,
Haotian

Ylei669, 2022-06-18:
I meet the same problem; my shape of feats and coords is **[17748, 5] and [17748, 4]**；I hope to get your advice,too! Thanks!

kentang-mit, 2022-06-18:
Hello @Ylei669 @zhouth-shiny,

Would you please specify the GPU, PyTorch, CUDA version that you are using? This is not a problem related to BEVFusion itself, but I will try finding out a solution together with you. The solution mentioned [here](https://github.com/mit-han-lab/bevfusion/issues/26) might be helpful.

Best,
Haotian

Ylei669, 2022-06-19:
Hi, @kentang-mit , I used the Tesla V100 , cuda version is 10.2 , pytorch 1.9.1.  I just tried the method of removing all "-gencode=arch=compute_XX,code=sm_XX "in the setup.py,but  it doesn't work. Thanks ！

kentang-mit, 2022-06-19:
Do you have a chance to check the `mmdetection` and `mmcv` versions on your machines? Let's first confirm that these versions match what I specified in the README. @Ylei669 @zhouth-shiny

Ylei669, 2022-06-19:
@kentang-mit Yes,I follow the instructions to install them。
Use the following commands for your reference：
pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu102/torch1.9.1/index.html
pip install mmdet==2.20.0
pip install numba
pip install nuscenes-devkit
apt-get install libopenmpi-dev
pip install mpi4py 
Thanks for your help!

kentang-mit, 2022-06-19:
Would you mind printing out the size of `indices` [here](https://github.com/mit-han-lab/bevfusion/blob/da93b27d0fe4563eecd46604e6cc5ddf1c5bd1ea/mmdet3d/ops/spconv/ops.py#L93)? It looks like the size is 0 and triggers the assertion error.

Ylei669, 2022-06-19:
Hi, @kentang-mit ,The size of indices is **[17748, 4]**, dtype=torch.int32; Thanks!

kentang-mit, 2022-06-19:
Currently I don't have very good solution out of my mind. I would suggest you to start from a clean Anaconda environment and try only installing dependencies specified in this repository. The error you encountered is related to the `getBlocks` call in [this file](https://github.com/mit-han-lab/bevfusion/blob/da93b27d0fe4563eecd46604e6cc5ddf1c5bd1ea/mmdet3d/ops/spconv/src/indice_cuda.cu), where some of the `numActIn`s or `numAct`s must be zero.

closed 2022-06-20 by zhouth-shiny

zzj403, 2022-07-29:
I've looked the mentioned issue #82 but I still can not fix the same problem.
Some times this error comes out, but sometimes it doesn't come out and just leave my training stucked, and the utilization of GPU are (GPU0 100% GPU1 0% GPU2 100% ...GPU7 100% )
Sometimes stuck at around  20th step, sometimes stuck in later epoch .
I've met this problem many times. Sad..

Deephome, 2022-09-16:
> 
hi, how did you solve this problem?

fdx24, 2023-07-31:
I have the same problem like this **N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0**, and `pip install spconv-cu113` is useful for me.