mit-han-lab/bevfusion #29: Errors occured when visualize the segmentation results of the fusion and camera-only model.

JoeyforJoy, 2022-06-18:
Thank you for your awesome work. 
When I try to visualize the segmentation results of the **fusion** and**camera-only**  baseline, some errors occured like following. 
However, there is **no error** when I run the LiDAR only seg model.
```bash
Traceback (most recent call last):
  File "tools/visualize.py", line 167, in <module>
    main()
  File "tools/visualize.py", line 88, in main
    outputs = model(**data)
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/ubuntu/repo/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 169, in forward
    feature = self.extract_camera_features(
  File "/home/ubuntu/repo/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 108, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/ubuntu/repo/bevfusion/mmdet3d/models/vtransforms/lss.py", line 76, in forward
    x = super().forward(*args, **kwargs)
  File "/home/ubuntu/anaconda3/envs/bev_fusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
TypeError: forward() takes 10 positional arguments but 11 were given
```

The script I run as following.
```bash
# fusion seg
export CONFIG=/home/ubuntu/repo/bevfusion/configs/nuscenes/seg/fusion-bev256d2-lss.yaml
export CHECKPOINT=/home/ubuntu/repo/bevfusion/pretrained/bevfusion-seg.pth
export OUTDIR=viz/bevfusion-seg
export MODE=pred

# # camera-only seg
# export CONFIG=/home/ubuntu/repo/bevfusion/configs/nuscenes/seg/camera-bev256d2.yaml
# export CHECKPOINT=/home/ubuntu/repo/bevfusion/pretrained/camera-only-seg.pth
# export OUTDIR=viz/camera-only-seg
# export MODE=pred

python tools/visualize.py ${CONFIG} --checkpoint ${CHECKPOINT} --mode ${MODE} --out-dir ${OUTDIR}
```

JoeyforJoy, 2022-06-18:
I found this error is due to the mismatch of the parameters in `BaseTransform`'s `forward()` function.
I create a PR #30 to fix this error.

kentang-mit, 2022-06-18:
Thank you for the contribution. The error is caused by the `metas` field in the forward function. I have approved your PR and merged it.

closed 2022-06-18 by kentang-mit