mit-han-lab/bevfusion #31: problem when data preparation

RouDanCongJii, 2022-06-19:
hi,
when i use `python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes` , 
there is the problem
`[                                                  ] 0/6008, elapsed: 0s, ETA:Traceback (most recent call last):
  File "tools/create_data.py", line 111, in <module>
    nuscenes_data_prep(
  File "tools/create_data.py", line 31, in nuscenes_data_prep
    nuscenes_converter.create_nuscenes_infos(
  File "C:\Users\Eric\Downloads\bevfusion-main\bevfusion-main\tools\data_converter\nuscenes_converter.py", line 98, in create_nuscenes_infos
    train_nusc_infos, val_nusc_infos = _fill_trainval_infos(
  File "C:\Users\Eric\Downloads\bevfusion-main\bevfusion-main\tools\data_converter\nuscenes_converter.py", line 191, in _fill_trainval_infos
    mmcv.check_file_exist(lidar_path)
  File "C:\Users\Eric\anaconda3\envs\pytorch\lib\site-packages\mmcv\utils\path.py", line 23, in check_file_exist
    raise FileNotFoundError(msg_tmpl.format(filename))
FileNotFoundError: file "./data/nuscenes\samples/LIDAR_TOP/n008-2018-08-01-16-03-27-0400__LIDAR_TOP__1533153857947444.pcd.bin" does not exist`

i use mini dataset of nuScenes and idk if this is the reason for this error.

environment:
python 3.8.12
pytorch 1.9.1
tqdm 4.64.0
torchpack 0.3.1
mmcv-full 1.4.0
mmdet3d 0.0.0
mmdet 2.20.0
nuscenes-devkit 1.1.9
cuda 11.7

thanks for your help

kentang-mit, 2022-06-19:
Hi @RouDanCongJii,

Yes, I believe that is the problem. The data preparation code is designed for  `v1.0-trainval` or `v1.0-test`.

Best,
Haotian

RouDanCongJii, 2022-06-20:
hi,

i download the test dataset and the mini dataset is the subset of full trainval dataset, so i would have thought it could be used. Actually  the subset of the trainval cannot be supported right?

kentang-mit, 2022-06-20:
Hi @RouDanCongJii,

I would suggest you to download the entire nuScenes dataset (including the map extension). We have no plan for supporting a subset of `v1.0-trainval`.

Best,
Haotian

closed 2022-06-20 by kentang-mit

RouDanCongJii, 2022-06-20:
thanks for your answers