mit-han-lab/bevfusion #32: cfg["sync_bn"]

Bincle-Yin, 2022-06-20:
Heip!!!!! what is cfg["sync_bn"]? Is it about distributed train?Can you tell me how to get it in cfg?

kentang-mit, 2022-06-20:
Yes, it is used for distributed training, but I remember that we did not specify it in our experiments. 

There are two ways you can use the `sync_bn` flag in the config. Simply applying 
```yaml
sync_bn: true
```
will convert all BN layers to SyncBN layers. If you write 
```yaml
sync_bn:
  exclude:
    - encoder
```
It means that all BN layers whose name in state_dict starts with `encoder` will not be converted to SyncBN.

Bincle-Yin, 2022-06-20:
thank you so much

kentang-mit, 2022-06-20:
No problem. I will temporarily close this issue.

closed 2022-06-20 by kentang-mit

Paul-Zhu, 2023-01-16:
what is the reason why did not use SyncBN ? Is it because of the pytorch version not support or when use SyncBN not improve precision so did not use?

Paul-Zhu, 2023-01-16:
hello,   what is the reason why did not use SyncBN ? Is it because of the pytorch version not support or when use SyncBN not improve precision so did not use? @kentang-mit