mit-han-lab/bevfusion #33: ObjectPaste class miss some details when sample_2d=True

songlilucky, 2022-06-20:
Thanks for your great work!
ObjectPaste class miss some details when sample_2d=True,
 `sampled_dict = self.db_sampler.sample_all(
                gt_bboxes_3d.tensor.numpy(),
                gt_labels_3d,
                gt_bboxes_2d=gt_bboxes_2d,
                img=img,
            )`
and `db_sampler` `def sample_all(self, gt_bboxes, gt_labels, img=None):`,  Could you update the `db_sampler` method for the baseline model?

kentang-mit, 2022-06-20:
Multi-modal copy-paste (i.e. MoCa or PointAugmenting) is not used in this work. You can assume that `sample_2d` is False.

songlilucky, 2022-06-20:
> Multi-modal copy-paste (i.e. MoCa or PointAugmenting) is not used in this work. You can assume that `sample_2d` is False.

thanks for your help!

closed 2022-06-20 by songlilucky