mit-han-lab/bevfusion #34: Usage With ROS2

robertokcanale, 2022-06-20:
Hello,
I would like to start using BEVFusion for a ROS2 project. I am quite interested in the visual and lidar features of the net. 

Would it be possible to showcase/discuss/have a tutorial on where BEVFusion can be configured with real sensors and real input streams? 

How could it be configured with real live data and what data could it give to us as an output? I know it is a general question but an overview at code level/additional tutorial on how to get data in and data out with a pre-trained model would be great!

Thank you in advance and hoping to start a discussion on the topic

kentang-mit, 2022-06-20:
Hi @robertokcanale,

Thank you for the interest! You might need to create a custom dataset file similar to [this one](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/nuscenes_dataset.py) for your own data. All fields returned in [this](https://github.com/mit-han-lab/bevfusion/blob/6847d45f65c7b59527a87df0b3f9f97f8fb3daac/mmdet3d/datasets/nuscenes_dataset.py#L274) `data` field are necessary, and you can get them from your sensors. Due to the distribution difference of real world datasets and nuScenes, our pretrained models might not work perfectly in your real use case. You probably will have to label some data and train BEVFusion by yourself. 

Best,
Haotian

kentang-mit, 2022-06-30:
Temporarily closed due to inactivity.

closed 2022-06-30 by kentang-mit