mit-han-lab/bevfusion #35: some question about Bev pool's pre computation.

AndyYuan96, 2022-06-20:
Hi, thanks for sharing the work, after reading the code, I have some question about the precomputation.
I see in bev pool code, you first calculate rank, then sort rank, then use cuda function to calculate sum, in cuda kernel function, after calculate the sum of points in grid, you just put the value to the corresponding position(calculated by grid position).
So I wonder, where is pre computation? what's the different between your bev pool and LSS's bev pool, except for the sum calculation? 
LSS's code:
final = torch.zeros((B, C, self.nx[2], self.nx[0], self.nx[1]), device=x.device)
final[geom_feats[:, 3], :, geom_feats[:, 2], geom_feats[:, 0], geom_feats[:, 1]] = x.

what's the difference between your pre computation and LSS's final[geom_feats[:, 3], :, geom_feats[:, 2], geom_feats[:, 0], geom_feats[:, 1]] = x.

kentang-mit, 2022-06-20:
Pre-computation is not released in this repo. It requires the extrinsic and intrinsic matrices to be unchanged across the entire dataset, but actually on nuscenes those parameters do change once several hundreds of frames. This will not influence the average latency, but taking care of such corner cases will make our code messy. If you are still interested in the specific implementation, I can talk about it in this thread, but it is just basically caching the geom_feats, ranks, etc. after the first time they are computed.

The major difference other than pre-computation is the interval reduction you see. LSS uses a prefix sum and keeps the results on the interval boundaries, while we perform interval reduction directly. We also save some intermediate DRAM activations by putting partial results in the GPU registers.

AndyYuan96, 2022-06-21:
> Pre-computation is not released in this repo. A simple implementation requires the extrinsic and intrinsic matrices to be unchanged across the entire dataset, but actually on nuscenes those parameters do change once several hundreds of frames. This will not influence the average latency, but taking care of such corner cases will make our code messy. If you are still interested in the specific implementation, I can talk about it in this thread, but it is just basically caching the geom_feats, ranks, etc. after the first time they are computed. You need to do dataloader-level modification to make sure the results are correct (this is the tricky part).
> 
> The major difference other than pre-computation is the interval reduction you see. LSS uses a prefix sum and keeps the results on the interval boundaries, while we perform interval reduction directly. We also save some intermediate DRAM activations by putting partial results in the GPU registers (as you can see, we do not have the DRAM tensor x in LSS).

I got it, I find your 2d augmentation param is fixed , so no augmentation for 2d image, so you did can just calculate once and use the same after, but whether no augmentation will have influence to performance? as LSS use true augmentation.