mit-han-lab/bevfusion #40: RuntimeError: CUDA error: no kernel image is available for execution on the device

lvxuanxuan123, 2022-06-24:
RuntimeError: CUDA error: no kernel image is available for execution on the device  
CUDA Kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.

显卡：4个显卡RTX 3090 Ti，每个显卡显存24GB
驱动：450
CUDA：11.1
torch:1.9.0+cu111
mmcv-full：1.4.0

kentang-mit, 2022-06-24:
Hi @lvxuanxuan123,

What about removing line 22 and 23 [here](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/setup.py#L22) and trying out the following command again?

```bash
python setup.py develop
```

Best,
Haotian

lvxuanxuan123, 2022-06-25:
> Hi @lvxuanxuan123,
> 
> What about removing line 22 and 23 [here](https://github.com/mit-han-lab/bevfusion/blob/86b077bea0fa401ba2b04bf1acf2e6c0ff0e53d8/setup.py#L22) and trying out the following command again?
> 
> ```shell
> python setup.py develop
> ```
> 
> Best, Haotian

Hi, I remove  "-gencode=arch=compute_70,code=sm_70","-gencode=arch=compute_75,code=sm_75" two lines.
And keep this line: "-gencode=arch=compute_86,code=sm_86", and the error still remain the same.

kentang-mit, 2022-06-25:
Please also try removing line 24. I'm not sure about the compute capability of RTX3090Ti since it is a very new product.

lvxuanxuan123, 2022-06-25:
> Please also try removing line 24. I'm not sure about the compute capability of RTX3090Ti since it is a very new product.

I tried remove line 24 just now, and it didn't work , the error still remain the same

kentang-mit, 2022-06-25:
I would suggest you to give us a clearer context. For example, what model are you evaluating? Are you able to run part of the network (at least those image backbone layers)? Also, you can start from the camera-only models and see whether the same error appears again.

kentang-mit, 2022-06-30:
Temporarily closed due to inactivity.

closed 2022-06-30 by kentang-mit