mit-han-lab/bevfusion #41: Cuda execution failed with error 2 ?

yt7589, 2022-06-24:
When I run tools/test.py it reported errors as bellow:

Traceback (most recent call last):
  File "tools/test.py", line 242, in <module>
    main()
  File "tools/test.py", line 205, in main
    outputs = single_gpu_test(model, data_loader)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/apis/test.py", line 17, in single_gpu_test
    result = model(return_loss=False, rescale=True, **data)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/mmcv/parallel/data_parallel.py", line 50, in forward
    return super().forward(*inputs, **kwargs)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/parallel/data_parallel.py", line 166, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 191, in forward
    feature = self.extract_lidar_features(points)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 128, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/models/backbones/sparse_encoder.py", line 119, in forward
    x = self.conv_input(input_sp_tensor)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/mnt/10t/yantao/software/anaconda3/envs/bev/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/ops/spconv/conv.py", line 163, in forward
    outids, indice_pairs, indice_pair_num = ops.get_indice_pairs(
  File "/mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/ops/spconv/ops.py", line 92, in get_indice_pairs
    return get_indice_pairs_func(
RuntimeError: /mnt/10t/yantao/zjad/anns/bevfusion/mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2

How to fix it?

kentang-mit, 2022-06-24:
Hi @yt7589,

Are you able to run inference on the camera-only models? Let's first confirm that the custom CUDA kernels will work on your device. After that, it might be an OOM problem, and potentially you can try out the solution I proposed [here](https://github.com/mit-han-lab/bevfusion/issues/26) related to switching the inference order of LiDAR and camera backbones.

Best,
Haotian

Trent-tangtao, 2022-06-28:
Hi, I met the same problem in training on 3090. There is no problem when I test it by `"torchpack dist-run -np 2 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox"`, but I meet the error when I train by `"torchpack dist-run -np 2 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml"`. 
[image]

kentang-mit, 2022-06-28:
Hi @Trent-tangtao,

Perhaps it is due to OOM. Currently I am not working on the support for training.

Best,
Haotian

kentang-mit, 2022-06-30:
Temporarily closed due to inactivity.

closed 2022-06-30 by kentang-mit

zjuliangxun, 2023-04-05:
I find this bug related to mismatched datatypes rather than memory issues. 
When I set model.train(), the bug will occur. However, with model.eval(), it will disappear. I stored the input indices of them and find that they are totally the same. To run the following piece, both of the input will fail.
```
from  mmdet3d.ops.spconv import  sparse_conv_ext
import torch as t
uu=t.randint(0, 1024, (15474, 4)).cuda()
# uu=t.load('static_obst/indices.pt')
sparse_conv_ext.get_indice_pairs_3d(uu,t.tensor(1).cuda(),[1024, 1024, 41],[1024, 1024, 41],[3, 3, 3], [1, 1, 1],[1, 1, 1],[1, 1, 1],[0, 0, 0],1,0)   
```
Then I find that as soon as I change the data type of t.tensor(1).cuda() to torch.int32, the bug is fixed.
To summary, this bug is because under the training mode, the parameter, batchsize, is of int64, and dismatches the cuda kernel's datatype.

xuwentai, 2023-05-24:
> May I ask which file you made the changes"change the data type of t.tensor(1).cuda() to torch.int32,"

zjuliangxun, 2023-05-25:
Sorry, but this change cannot fix the bug at all. It appears again after I commit this issue……

从 Windows 版邮件发送

发件人: xuwentai
发送时间: 2023年5月24日 15:06
收件人: mit-han-lab/bevfusion
抄送: xunliang; Comment
主题: Re: [mit-han-lab/bevfusion] Cuda execution failed with error 2 ?(Issue #41)

May I ask which file you made the changes"change the data type of t.tensor(1).cuda() to torch.int32,"

heli223, 2023-12-22:
i met the same problem. Actually, I have train detection successfully before, and today i train segementation, it occurs. More strangely， the train for detection fails.

**bevfusion/mmdet3d/ops/spconv/src/indice_cuda.cu 124
cuda execution failed with error 2**