mit-han-lab/bevfusion #43: Why is this variable lidar_aug_matrix not used from points to cur_coords

study1994, 2022-06-25:
Hello,When I read the code, I was confused from [points](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/base.py#L246) to [cur_coords](https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/base.py#L265)  use cur_lidar2image->cur_img_aug_matrix instead of  lidar_aug_matrix->cur_lidar2image->cur_img_aug_matrix。Because I want to print the 3D box and display it on the picture with 256, 704 but failed。

kentang-mit, 2022-06-25:
You are right. We do have lidar_aug_matrix in the first during training. I forgot to add that line of code to the current version because it does not affect influence. By the way, if you are interested in visualization, take a look at [this file](tools/visualize.py). We directly provide a visualization interface to display 3D boxes on the original 1600x900 images.

study1994, 2022-06-25:
Thank you for your reply, Did you train first Modality C and Modality L then finetuning just like transfusion when training nuscenes with Modality C+L。or i can train it directly with the following commands after use the above mentioned `lidar_aug_matrix`:
```
CUDA_VISIBLE_DEVICES="4,5,6,7" torchpack dist-run -np 4 python tools/train.py \
        configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml \
        --run_dir runs/convfuser \
        2>&1 | tee runs/convfuser/train.txt
```
If both the point cloud and the image are enhanced with data by aug_matrix，Are they not in the same space？

kentang-mit, 2022-06-25:
In our paper we use the fine-tuning setting. Point clouds and lifted image features are guaranteed to be aligned in the same space in this work, details can be found in our view transformer implementation.

We will start to answer questions related to training details until we release the full training code, please stay tuned.

closed 2022-06-25 by kentang-mit