mit-han-lab/bevfusion #44: A Request to release detection result file on test/val set.

lxy-04, 2022-06-25:
Hi! BEVFusion-e is an excellent job. The algorithm has the highest NDS on the 'nuScenes' dataset. At the same time, good detection algorithms are very important for other tasks (Like 'Tracking'). Therefore, if you can release the detection results of the algorithm on the nuscenes 'test' set and 'valid' set, it will be very helpful for tracking and other work. looking forward to your reply!:grinning:

kentang-mit, 2022-06-26:
Hi @lxy-04,

Thank you for your interest. You can obtain the predictions of BEVFusion on the validation dataset using this repo. If you are interested in ensemble results on val, we do not have the bandwidth to re-generate them at the current moment, but feel free to shoot us an Email if some part of your potential research paper depends on our results.

Best,
Haotian

kentang-mit, 2022-06-26:
We will contact through Email. Close it for now.

closed 2022-06-26 by kentang-mit