mit-han-lab/bevfusion #45: Inference with camera-only model

pycoco, 2022-06-26:
Any config  to correct inference camera-only model? where should i change in config. Firstly, i test without any change, but i met below error.
[image]
Next i disable LoadBEVSegmentation &  gt_masks_bev, there is another error in bevpooling.
[image]

pycoco, 2022-06-27:
After i fixed problem of "location" key error. And i use original config to inference. As a result , I find that all the points are been filtered. 
[image]
The code in https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/base.py#L156
so, i met error
[image]
Any solutions to fixed it?

kentang-mit, 2022-06-27:
Hi @pycoco,

This did not happen during my evaluation. May I ask whether you are using the full `v1.0-trainval` dataset or the `v1.0-mini` subset? Also, did you download the nuScenes map extension?

Best,
Haotian

pycoco, 2022-06-27:
@kentang-mit  I use v1.0-mini and  i also download the nuScenes map extension.

pycoco, 2022-06-27:
> @kentang-mit I use v1.0-mini and i also download the nuScenes map extension.

Also, i test with fusion model. 
`torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`

kentang-mit, 2022-06-27:
Hi @pycoco,

We currently only support the full `v1.0-trainval` (or `v1.0-test`) set. I haven't tested our code on `v1.0-mini`. Perhaps you will have to download the full dataset.

Best,
Haotian

pycoco, 2022-06-27:
> Hi @pycoco,
> 
> We currently only support the full `v1.0-trainval` (or `v1.0-test`) set. I haven't tested our code on `v1.0-mini`. Perhaps you will have to download the full dataset.
> 
> Best, Haotian

Thanks, i already locate the problem. It caused by pytorch.

closed 2022-06-27 by pycoco

SuperHenry2333, 2022-07-03:
> > Hi @pycoco,
> > We currently only support the full `v1.0-trainval` (or `v1.0-test`) set. I haven't tested our code on `v1.0-mini`. Perhaps you will have to download the full dataset.
> > Best, Haotian
> 
> Thanks, i already locate the problem. It caused by pytorch.

Hi @pycoco ,
I met the same problem. Could do tell me your solution? Thank you very much!

bsaisudh, 2022-07-18:
Hi @pycoco ,

I tried to run evaluation for detection variant as mentioned in the readme and I am also faceing the same problem with 'Location' key error. 

`File "/bevfusion/mmdet3d/datasets/nuscenes_dataset.py", line 218, in get_data_info
    location=info["location"],
KeyError: 'location'`

I tried with both full and mini data datasets and the issue still persists. How did you solve the error?

bsaisudh, 2022-07-26:
> Hi @pycoco ,
>
> I tried to run evaluation for detection variant as mentioned in the readme and I am also faceing the same problem with 'Location' key error.
>
> `File [...]

I was running create_data.py from mmdet3d repo rather than bevfusion repo. This was the problem and solved it.

304886938, 2022-10-18:
Hi @bsaisudh ,
Can you tell me which version of mmdet3d you used to solve the problems?
Thank you for your answer!

heli223, 2023-10-25:
> > Hi @pycoco,
> > We currently only support the full `v1.0-trainval` (or `v1.0-test`) set. I haven't tested our code on `v1.0-mini`. Perhaps you will have to download the full dataset.
> > Best, Haotian
> 
> Thanks, i already locate the problem. It caused by pytorch.

@pycoco hi, can you tell me how you fixed the location problem? I have the same issue with the following command：
torchpack dist-run -np 1 python tools/test.py configs/nuscenes/seg/fusion-bev256d2-lss.yaml ./tools/pretrained/bevfusion-seg.pth --eval map --out map_bevfusion-det.pkl 

 File "tools/test.py", line 230, in <module>
    main()
  File "tools/test.py", line 203, in main
    outputs = multi_gpu_test(model, data_loader, args.tmpdir, args.gpu_collect)
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmdet/apis/test.py", line 96, in multi_gpu_test
    for i, data in enumerate(data_loader):
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/_utils.py", line 434, in reraise
    raise exception
KeyError: Caught KeyError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/xx/bevfusion-heli/bevfusion-main/mmdet3d/datasets/custom_3d.py", line 291, in __getitem__
    return self.prepare_test_data(idx)
  File "/home/xx/bevfusion/bevfusion-main/mmdet3d/datasets/custom_3d.py", line 182, in prepare_test_data
    example = self.pipeline(input_dict)
  File "/home/xx/anaconda3/envs/bevfusion/lib/python3.8/site-packages/mmdet/datasets/pipelines/compose.py", line 41, in __call__
    data = t(data)
  File "/home/xx/bevfusion/bevfusion-main/mmdet3d/datasets/pipelines/loading.py", line 294, in __call__
    location = data["location"]
KeyError: 'location'