mit-han-lab/bevfusion #47: 执行 python setup.py develop报错

zhuxuanze456, 2022-06-28:
你好，
我想咨询一下关于执行python setup.py develop报错的问题，报错的显示如下：
building 'mmdet3d.ops.spconv.sparse_conv_ext' extension
gcc -pthread -B /root/anaconda3/envs/bevFusion/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DWITH_CUDA -I/home/d50024694/bevfusion-main/mmdet3d/ops/spconv/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/TH -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-10.1/include -I/root/anaconda3/envs/bevFusion/include/python3.7m -c mmdet3d/ops/spconv/src/all.cc -o build/temp.linux-x86_64-cpython-37/mmdet3d/ops/spconv/src/all.o -w -std=c++14 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=sparse_conv_ext -D_GLIBCXX_USE_CXX11_ABI=0
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
gcc -pthread -B /root/anaconda3/envs/bevFusion/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DWITH_CUDA -I/home/d50024694/bevfusion-main/mmdet3d/ops/spconv/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/TH -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-10.1/include -I/root/anaconda3/envs/bevFusion/include/python3.7m -c mmdet3d/ops/spconv/src/indice.cc -o build/temp.linux-x86_64-cpython-37/mmdet3d/ops/spconv/src/indice.o -w -std=c++14 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=sparse_conv_ext -D_GLIBCXX_USE_CXX11_ABI=0
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
/usr/local/cuda-10.1/bin/nvcc -DWITH_CUDA -I/home/d50024694/bevfusion-main/mmdet3d/ops/spconv/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/TH -I/root/anaconda3/envs/bevFusion/lib/python3.7/site-packages/torch/include/THC -I/usr/local/cuda-10.1/include -I/root/anaconda3/envs/bevFusion/include/python3.7m -c mmdet3d/ops/spconv/src/indice_cuda.cu -o build/temp.linux-x86_64-cpython-37/mmdet3d/ops/spconv/src/indice_cuda.o ### -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -w -std=c++14 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=sparse_conv_ext -D_GLIBCXX_USE_CXX11_ABI=0

### **nvcc fatal   : Unsupported gpu architecture 'compute_86'
error: command '/usr/local/cuda-10.1/bin/nvcc' failed with exit code 1**
环境信息：
python:3.8
cuda:10.1
torch:torch-1.8.1+cu101
torchvision:torchvision-0.9.1+cu101
mmcv= 1.4.0
mmdetection = 2.20.0

想问一下这种情况是因为cuda版本较低的原因吗？？cuda有没有最低版本限制？

zhuxuanze456, 2022-06-28:
@kentang-mit Thank you. Compilation succeeded.

kentang-mit, 2022-06-28:
That's great. I'll temporarily close this issue.

closed 2022-06-28 by kentang-mit