mit-han-lab/bevfusion #51: Problem about reading the code

mc171819, 2022-06-30:
Hi, bevfusion is a wonderful work. However, I am a new user of mmdet3d and I am confused about how your project read the config.  I found config file in mmdet3d official code is a single .py file, but in your example you just load the convfuse.yaml but there are many default.yaml seem also necessary, so how the default.yaml be read?

By the way, it will be nice for you to produce an OpenPCDet version of BEVfusion!
@kentang-mit

mc171819, 2022-06-30:
I also wonder have you tried using CNN 2d backbone and dense head without transformer? I found you are using swin transformer to extract 2d features and Transfuion head, which is also a transformer-based dense head for 3d detection, is used. Is it because only transformer based structure can be combined successfully?

kentang-mit, 2022-06-30:
Hi @mc171819,

For your first question, the YAML configuration systems use `default.yaml` at higher level of directory hierarchy as the base configuration file. Configurations in the deeper folders will **inherit** and **modify** the base `default.yaml` configs. So yes, your understanding is correct, all the config files are necessary. The advantage of this YAML configuration system is that it can avoid duplicate code snippets when you are doing a lot of experiments with minor differences in hyper-parameters (especially when these hyperparameters are nested in the dataloader configurations, like [here](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/configs/nuscenes/default.yaml#L120)).

For OpenPCDet reproduction, we will probably give it a try once we have more bandwidth.

Related to transformers, we don't have to use them in BEVFusion. We have ablation studies on ResNet50 for the image backbone, which achieves slightly worse performance than SwinTransformer. For the head, we actually use center-based heads for all BEV segmentation experiments, and it will also work for the detection experiments.

Best,
Haotian

mc171819, 2022-07-01:
> Hi @mc171819,
>
> For your first question, the YAML configuration systems use `default.yaml` at higher level of directory hierarchy as the base configuration file. Configurations in the deeper folders [...]

Thank you for your reply! In my understanding, camera is transform into bev style in self.encoders["camera"]["vtransform"], which is the new designed bevpooling module, is that correct?

kentang-mit, 2022-07-01:
That's partially correct. We also have pre-computation but not released in this repo. Besides, if you look at the best model we provided, actually we added a 2.5D depth stream to the view transformer, this is not part of LSS. It has some improvement but also brings about latency overhead.

mc171819, 2022-07-01:
> 

Thank you for your patient answer, I'll close this issue.

mc171819, 2022-07-04:
Hi, after learning something about LSS, in my understanding, in the released version of bevfusion, the main difference between LSS and bevfusion of bev pooling is that you provide a new function named QuickCumsumCuda in mmdet3d/ops/bev_pool/bev_pool.py rather than using QuickCumsum, which is used by LSS?

kentang-mit, 2022-07-04:
Yes.

closed 2022-07-04 by mc171819

wulinaaa, 2023-08-19:
> I also wonder have you tried using CNN 2d backbone and dense head without transformer? I found you are using swin transformer to extract 2d features and Transfuion head, which is also a [...]

Hello, May I know where the swing transformer  to extract 2d features code is, I can't find it.