mit-han-lab/bevfusion #52: The geometry transform of image_to_lidar

Trent-tangtao, 2022-06-30:
Hi, 
Why you use cam_to_ego  and  ego_to_lidar in [get_geometry](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/mmdet3d/models/vtransforms/base.py#L79) to achieve image_to_lidar transformation, not just use the inverse of lidar2image in [lidar2image](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/mmdet3d/datasets/nuscenes_dataset.py#L260)

kentang-mit, 2022-06-30:
Hi @Trent-tangtao,

Although I haven't tried that, I believe it should be correct if you do so.

Best,
Haotian

Trent-tangtao, 2022-07-01:
Thanks for your kind reply!

Trent-tangtao, 2022-07-04:
Hi,
why the RT [extra_rots and extra_trans](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/mmdet3d/models/vtransforms/base.py#L239) of lidar aug not passed to [get_geometry](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/mmdet3d/models/vtransforms/base.py#L278) BaseDepthTransform ?

kentang-mit, 2022-07-04:
See [here](https://github.com/mit-han-lab/bevfusion/issues/43). I just forgot them during the reformatting process. Will add in the future.

closed 2022-07-04 by Trent-tangtao