mit-han-lab/bevfusion #53: ImportError: cannot import name 'recursive_eval' from 'mmdet3d.utils'

Spring-lovely, 2022-06-30:
Hi,
ImportError: cannot import name 'recursive_eval' from 'mmdet3d.utils'
What can I do?

kentang-mit, 2022-06-30:
You probably need to run

```bash
python setup.py develop
```
under the root folder. This file is not part of the original `mmdetection3d`.

kentang-mit, 2022-06-30:
We have requirements on the `mmcv` and `mmdetection` version. Please read the README first.

Spring-lovely, 2022-07-01:
Hi,
Thanks!
now I can run test.py about detection
[image]
 but when I run `python tools/visualize.py` errer:
[image]
I want to visualize the effect after object detection, how should I do it?

kentang-mit, 2022-07-01:
Would you mind trying out this command?

```bash
torchpack dist-run -np 1 python tools/visualize.py [config path] --checkpoint [checkpoint path] --mode pred
```

By the way, your evaluation results don't seem familiar to me. I'm not sure whether it is on `v1.0-mini`.

Spring-lovely, 2022-07-01:
Yes，I use v1.0-mini.
but I run your command.error:No module named 'torchpack.utils.tqdm'----i sure, my env has torchpack .
[image]

kentang-mit, 2022-07-01:
My suggestion will be just to replace it with

```python
from tqdm import tqdm
```

By the way, I have to mention that I never encountered this problem after installing torchpack. Maybe @zhijian-liu has more insights on that.

kentang-mit, 2022-07-07:
I'll temporarily close this due to inactivity. Feel free to reopen if you have further questions.

closed 2022-07-07 by kentang-mit

hitbuyi, 2024-04-18:
I have the same issue,the reason is that recursive_val is funciton defined  under project direction, and python search it in lib directory, as below 
```
(pt110) hitbuyi@hitbuyi-Dell-G15-5511:~/PycharmProjects/Pytorch_Project/BEVFusion_MIT$ torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
Traceback (most recent call last):
  File "tools/test.py", line 19, in <module>
    from mmdet3d.utils import recursive_eval
ImportError: cannot import name 'recursive_eval' from 'mmdet3d.utils' (/home/hitbuyi/.conda/envs/pt110/lib/python3.8/site-packages/mmdet3d-0.17.1-py3.8-linux-x86_64.egg/mmdet3d/utils/__init__.py)
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[32825,1],0]
  Exit code:    1
--------------------------------------------------------------------------
```
what you should do is to make python search recursive_val from project directory rather than lib directory