mit-han-lab/bevfusion #55: CUDA error: initialization error when calling `cusparseCreate(handle)`

hongjiacheng1014, 2022-07-01:
Hi. There is some problems about the evaluation.
[image]
How to fix it?

kentang-mit, 2022-07-01:
Hi @hongjiacheng1014,

Thanks for reporting the problem. Would you mind providing more details about your experiments? For example, what are the GPU and library versions you are using? Is it possible for you to run camera-only / LiDAR-only models? 

Best,
Haotian

hongjiacheng1014, 2022-07-02:
Hi @kentang-mit,
Thanks for your help. There is some information about GPU. It seems GPU OOM.
GPU: 2 NVIDIA GeForce GTX 1080 Ti (11264MB)
No. I cannot run camera-only and LiDAR-only models.

kentang-mit, 2022-07-02:
Got it. Have you ever tried commenting out line 22-25 [here](https://github.com/mit-han-lab/bevfusion/blob/cb6cd789d1e93e160052c7305f8c9e9407b9a954/setup.py#L22)? I will try to reproduce the OOM problem on a machine with smaller GPU memory.

hongjiacheng1014, 2022-07-05:
Thanks for your help, @kentang-mit. I solved this problem by freeing some GPU memory.

kentang-mit, 2022-07-05:
Good to know. I'll close this for now.