mit-han-lab/bevfusion #59: bevfusion-det error result

you-old, 2022-07-07:
hi boss, use command:
```
# # fusion det
export CONFIG=./configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml
export CHECKPOINT=./pretrained/bevfusion-det.pth
export OUTDIR=viz/fusion-det
export MODE=pred

MASTER_HOST=localhost:1331 python tools/visualize.py ${CONFIG} --checkpoint ${CHECKPOINT} --mode ${MODE} --out-dir ${OUTDIR}
```

the result :
[image]
[image]

why ???

kentang-mit, 2022-07-11:
Closed due to inactivity.

closed 2022-07-11 by kentang-mit

RouDanCongJii, 2022-10-04:
hi,

which configuration files should i modify to adjust the nms_threshold?

Zhanfury, 2023-12-06:
hi,
I ‘ve tried the visualization code you provided，but the following error confused me：
“Traceback (most recent call last):
  File "tools/visualize.py", line 166, in <module>
    main()
  File "tools/visualize.py", line 88, in main
    outputs = model(**data)
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward
    output = self.module(*inputs[0], **kwargs[0])
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward
    outputs = self.forward_single(
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func
    return old_func(*args, **kwargs)
  File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single
    feature = self.extract_camera_features(
  File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features
    x = self.encoders["camera"]["vtransform"](
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/zfy/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward
    x = super().forward(*args, **kwargs)
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
  File "/home/zfy/bevfusion/mmdet3d/models/vtransforms/base.py", line 356, in forward
    x = self.get_cam_feats(img, depth, mats_dict)           #????
  File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func
    return old_func(*args, **kwargs)
TypeError: get_cam_feats() takes 3 positional arguments but 4 were given”
I would appreciate it if you could give me some suggestions

971022jing, 2024-01-03:
> hi, I ‘ve tried the visualization code you provided，but the following error confused me： “Traceback (most recent call last): File "tools/visualize.py", line 166, in main() File "tools/visualize.py", line 88, in main outputs = model(**data) File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl return forward_call(*input, **kwargs) File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/parallel/distributed.py", line 886, in forward output = self.module(*inputs[0], **kwargs[0]) File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl return forward_call(*input, **kwargs) File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func return old_func(*args, **kwargs) File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 253, in forward outputs = self.forward_single( File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 98, in new_func return old_func(*args, **kwargs) File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 301, in forward_single feature = self.extract_camera_features( File "/home/zfy/bevfusion/mmdet3d/models/fusion_models/bevfusion.py", line 133, in extract_camera_features x = self.encoders["camera"]["vtransform"]( File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl return forward_call(*input, **kwargs) File "/home/zfy/bevfusion/mmdet3d/models/vtransforms/depth_lss.py", line 100, in forward x = super().forward(*args, **kwargs) File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func return old_func(*args, **kwargs) File "/home/zfy/bevfusion/mmdet3d/models/vtransforms/base.py", line 356, in forward x = self.get_cam_feats(img, depth, mats_dict) #???? File "/home/zfy/.conda/envs/bevfusion/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 186, in new_func return old_func(*args, **kwargs) TypeError: get_cam_feats() takes 3 positional arguments but 4 were given” I would appreciate it if you could give me some suggestions

Have you found a solution to this problem?
I still have the same problem.