mit-han-lab/bevfusion #60: how can I dump a bevfusion model to onnx?

hych2020, 2022-07-07:
I try to dump a detection model to onnx by following command, but failed

```
python tools/export.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth
```

kentang-mit, 2022-07-07:
Hi @hych2020,

It might be hard to export BEVFusion to onnx currently. The input problem you met with can be solved by manually passing these arguments into the forward function instead of relying on the containers in `mmcv`. However, our model is composed of SparseConv layers and customized CUDA ops (such as BEV pooling). Exporting them to onnx will require significant work and we haven't started trying that. 

Best,
Haotian

hych2020, 2022-07-08:
Hi @kentang-mit 

Thank you.

I got it.

So How do you measure FPS on Specific hardware， using the pytorch model directly?
I thought it was converting to onnx first，then deployed with TensorRT。

kentang-mit, 2022-07-08:
Hi @hych2020,

You can check out [this issue](https://github.com/mit-han-lab/bevfusion/issues/14). TensorRT is not used in latency measurement. With that being said, I expect the latency to be lower than what is reported in the paper if we eventually have the pipeline you mentioned.

Best,
Haotian

hych2020, 2022-07-08:
OK! Thank you very much!

closed 2022-07-08 by hych2020

RangineniBalaji, 2024-07-08:
> I try to dump a detection model to onnx by following command, but failed
> 
> ```
> python tools/export.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth
> ```
> 
> error [image]
> 
> after fix the config problem, another error occurred [image]

Hi hych, do you exported the model to onnx.?