mit-han-lab/bevfusion #62: How do I train the model？

Fengshihao1, 2022-07-08:
Hi boss，
  Thank you very much for your work！I want to try BEVFusion to train a model, but I don't have a full configfile, can you provide a full profile? Thank you very much.

kentang-mit, 2022-07-08:
Training details will be released in the future. Please stay tuned.

closed 2022-07-08 by kentang-mit

Fengshihao1, 2022-10-11:
Thank you very much for your answer, I will very much look forward to your training details.

HuangVictorAuto, 2022-10-12:
> Training details will be released in the future. Please stay tuned.

@kentang-mit , have you released the training details already? 
I have seen the training for camera_only detection, segmentation, lidar only detection, segmentation.  Will there be a config to train the multi task and fusion model all in one? thanks!