mit-han-lab/bevfusion #65: when i run text conmmand ,The process is stuck in the loading data section

baohaooo, 2022-07-09:
when i run text conmmand ,The process is stuck in the loading data section。I tested the algorithm on the val data set. Normally, 6019 should be loaded, but I loaded 6021, and it stuck here, and no errors were reported. How can I solve this problem. The last line of terminal display information is: 
self post_ center_ range = torch. tensor(
[>>] 6021/6019, 6.4 task/s, elapsed: 936s, ETA: 0spacket_ write_ wait: Conne
[image]

kentang-mit, 2022-07-10:
Hi @baohaooo,

Is it possible for you to try out running evaluation on 1 GPU and see whether the same problem occurs again? Thanks!

Best,
Haotian

baohaooo, 2022-07-10:
Hi sir 
       The current situation is: 1 Yesterday, after waiting for about 6 or 7 hours, the program automatically continued to run, and I finally got the test result. 2. Last night, Beijing time, I ran this command “CUDA_ VISIBLE_ Devices=5,6,8 torchpack dist run -np 3 Python "tools/test.py configurations/scenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml trained/bevfusion-det.pth --eval bbox --eval options" jsonfile_prefix=./results.json "”. I used three rtx3090 to test, but now seven hours have passed still stuck at 6021/6019. Should I use only one GPU for testing?
       Thx.

Solacex, 2022-07-11:
Hello,

I think we met the same problem. When training/evaluating with multiple cards, the utilization rate of one card drops to 0, and the program is stuck there and cannot be continued. 

Is there any suggestions for debugging this ?

baohaooo, 2022-07-11:
I have solved this problem. This situation will not occur with only one GPU

kentang-mit, 2022-07-11:
Hi @Solacex @baohaooo,

Thanks for the update. I believe that one way to debug is to add a `-v` flag to your command like:

```bash
torchpack dist-run -np 8 -v python ...
```

where `...` is the rest of your command. Torchpack suppresses error messages from all GPUs other than GPU0, and the `-v` flag can restore those error messages. Hopefully you will get more information from that.

Best,
Haotian

Solacex, 2022-07-12:
Thanks for your response, I don't have any other questions.

kentang-mit, 2022-07-12:
OK, I'll close it for now.

ChongjianGE, 2022-11-29:
> ```shell
> -v
> ```

Hi, 

I think I met the same question as you. The utilization rate of one card drops to 0, and the programs seem to be stuck. Did you solve the problems? 

Best