mit-han-lab/bevfusion #69: Question abouth the sparse encoder

Solacex, 2022-07-12:
Hello, 

Thanks for your inspiring work, and I am trying to run the code you provided. 

I met the following error, which indicates the shape mismatch in the sparse encoder. I have scrutinized the shape of all LiDAR scans in the NuScenes but cannot find any error.

Any suggestions for solving this? 

Traceback (most recent call last):
  File "tools/train.py", line 89, in <module>
    main()
  File "tools/train.py", line 78, in main
    train_model(
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/apis/train.py", line 128, in train_model
    runner.run(data_loaders, [("train", 1)])
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 127, in run
    epoch_runner(data_loaders[i], **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/runner/epoch_based_runner.py", line 15, in train
    super().train(data_loader, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 50, in train
    self.run_iter(data_batch, train_mode=True, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/runner/epoch_based_runner.py", line 29, in run_iter
    outputs = self.model.train_step(data_batch, self.optimizer,
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/parallel/distributed.py", line 52, in train_step
    output = self.module.train_step(*inputs[0], **kwargs[0])
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/models/fusion_models/base.py", line 78, in train_step
    losses = self(**data)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 194, in forward
    feature = self.extract_lidar_features(points)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/models/fusion_models/bevfusion.py", line 128, in extract_lidar_features
    x = self.encoders["lidar"]["backbone"](feats, coords, batch_size, sizes=sizes)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/mmcv/runner/fp16_utils.py", line 128, in new_func
    output = old_func(*new_args, **new_kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/models/backbones/sparse_encoder.py", line 122, in forward
    x = encoder_layer(x)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/ops/spconv/modules.py", line 132, in forward
    input = module(input)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/mmdet3d/ops/spconv/modules.py", line 136, in forward
    input.features = module(input.features)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/modules/batchnorm.py", line 168, in forward
    return F.batch_norm(
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/functional.py", line 2280, in batch_norm
    _verify_batch_size(input.size())
  File "/root/paddlejob/workspace/whole/bevfusion-main/env/guangrui_cuda11/lib/python3.8/site-packages/torch/nn/functional.py", line 2248, in _verify_batch_size
    raise ValueError("Expected more than 1 value per channel when training, got input size {}".format(size))
ValueError: Expected more than 1 value per channel when training, got input size torch.Size([1, 32])

kentang-mit, 2022-07-12:
Looks like this is a training-related problem. Maybe you can wait until we release full training details. To me, it seems like some BN1d layer has batch size = 1 during training, which is definitely not allowed.

Solacex, 2022-07-12:
Thanks for your response, and look forward the full training code.

closed 2022-07-12 by Solacex