mit-han-lab/bevfusion #74: Why remove "get_geometry" out of forward process could not gain acceleration?

cherry-yh, 2022-07-13:
Hi, as you mentioned in your paper, "precomputation" of camera feature point cloud coordinates in advance, record grid indices and rank of each point by "caching mechanism". I guess, during inference, all the variables do not need to compute in realtime while directly load from cache.
Seems your code has not released this part, and we assume the cam intrinsics and extrinsics are fixed (though nuScenes is not fixed), we move the get_geometry to the initialization part, during inference, we just load the initialized variable, and test infer speed again. But why the infer speed is not improved?  ("from 17ms to 4ms" as mentioned in your paper).

kentang-mit, 2022-07-13:
There are actually more things that you can cache. For example, line 86-92 [here](https://github.com/mit-han-lab/bevfusion/blob/98319eeaeb3fefa9320069a364aaf669cd8dc1c2/mmdet3d/ops/bev_pool/bev_pool.py#L86).

cherry-yh, 2022-07-15:
Thank you for your reply.
As your suggestions, I tried to move the following items out of forward process to __init__ part as Class members:
get_geometry;
ranks calculation, and ranks sort;
geom_feats voxelization;
kept mask;

during forward, all the above variables do not need to recalculate and just load from Class members.
But the infer speed is still not improved, it is a bit weird. 
Thanks anyway, if your code of this part is released that will be helpful~~~

kentang-mit, 2022-07-16:
I think your implementation should be correct. How did you measure the latency? [Here](https://github.com/mit-han-lab/bevfusion/issues/14) is the way we obtain the numbers.