mit-han-lab/bevfusion #77: Coordinate system conversion problem

chdzhen, 2022-07-17:
in mmdet3d/models/vtransforms/base.py.  
def get_geometry()

line 99 :
# cam_to_ego
        points = torch.cat(
            (
                points[:, :, :, :, :, :2] * points[:, :, :, :, :, 2:3],
                points[:, :, :, :, :, 2:3],
            ),
            5,
        )
        combine = rots.matmul(torch.inverse(intrins))
        points = combine.view(B, N, 1, 1, 1, 3, 3).matmul(points).squeeze(-1)
        points += trans.view(B, N, 1, 1, 1, 3)
# ego_to_lidar
        points -= lidar2ego_trans.view(B, 1, 1, 1, 1, 3)
        points = (
            torch.inverse(lidar2ego_rots)
            .view(B, 1, 1, 1, 1, 3, 3)
            .matmul(points.unsqueeze(-1))
            .squeeze(-1)
        )
line 117 

cam_to_ego and ego_to_lidar  is not one ego!!!  
should use came2lidar , data["lidar2camera"] have used one ego!!!   this have convert in nuscenes_converter.py

kentang-mit, 2022-07-17:
I did not quite understand what you are talking about. Would you mind giving a concrete example?

chdzhen, 2022-07-18:
because the ego is move, so cam_to_lidar should cam->ego_1->global->ego_0->lidar.  
like: tools/data_converter/nuscenes_converter.py line337:    # sweep->ego->global->ego'->lidar

kentang-mit, 2022-07-18:
I don't think we need to that because we only use camera images from one frame.

chdzhen, 2022-07-18:
in NuscenesData,  lida_data、camera images in one frame is not acquired at the same time;

kentang-mit, 2022-07-18:
Yes, I understand your point, and this is pretty much a problem for all existing datasets. You are welcomed to have a try on your idea, but I don't think that will impact the final results because the timestamp differences between camera and LiDAR is really small.

chdzhen, 2022-07-19:
thank you, you did a great job

kentang-mit, 2022-07-19:
You're welcome. I will temporarily close this issue since it is resolved.

closed 2022-07-19 by kentang-mit