mit-han-lab/bevfusion #79: ERROR in Evaluation

zccjjj, 2022-07-20:
Hello, I think I meet trouble and need help urgently! 
When I run 'test.py' as what the Readme said, I encountered this error, and the terminal was stuck:
'(bevfusion) root@zhangcaiji:~/bevfusion# torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
Config (path: configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml): {'seed': 0, 'deterministic': False, 'checkpoint_config': {'interval': 1, 'max_keep_ckpts': 1}, 'log_config': {'interval': 50, 'hooks': [{'type': 'TextLoggerHook'}, {'type': 'TensorboardLoggerHook'}]}, 'load_from': None, 'resume_from': None, 'cudnn_benchmark': False, 'fp16': {'loss_scale': {'growth_interval': 2000}}, 'max_epochs': 20, 'runner': {'type': 'CustomEpochBasedRunner', 'max_epochs': 20}, 'dataset_type': 'NuScenesDataset', 'dataset_root': 'data/nuscenes/', 'gt_paste_stop_epoch': -1, 'reduce_beams': 32, 'load_dim': 5, 'use_dim': 5, 'load_augmented': None, 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0], 'voxel_size': [0.075, 0.075, 0.2], 'image_size': [256, 704], 'augment2d': {'resize': [[0.48, 0.48], [0.48, 0.48]], 'rotate': [0.0, 0.0], 'gridmask': {'prob': 0.0, 'fixed_prob': True}}, 'augment3d': {'scale': [1.0, 1.0], 'rotate': [0.0, 0.0], 'translate': 0.0}, 'object_classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'map_classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider'], 'input_modality': {'use_lidar': True, 'use_camera': True, 'use_radar': False, 'use_map': False, 'use_external': False}, 'train_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ObjectPaste', 'stop_epoch': -1, 'db_sampler': {'dataset_root': 'data/nuscenes/', 'info_path': 'data/nuscenes/nuscenes_dbinfos_train.pkl', 'rate': 1.0, 'prepare': {'filter_by_difficulty': [-1], 'filter_by_min_points': {'car': 5, 'truck': 5, 'bus': 5, 'trailer': 5, 'construction_vehicle': 5, 'traffic_cone': 5, 'barrier': 5, 'motorcycle': 5, 'bicycle': 5, 'pedestrian': 5}}, 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'sample_groups': {'car': 2, 'truck': 3, 'construction_vehicle': 7, 'bus': 4, 'trailer': 6, 'barrier': 2, 'motorcycle': 6, 'bicycle': 6, 'pedestrian': 2, 'traffic_cone': 2}, 'points_loader': {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32}}}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.48, 0.48], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [0.0, 0.0], 'rand_flip': False, 'is_train': True}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [1.0, 1.0], 'rot_lim': [0.0, 0.0], 'trans_lim': 0.0, 'is_train': True}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'data/nuscenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'RandomFlip3D'}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectNameFilter', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'ImageNormalize', 'mean': [103.53, 116.28, 123.675], 'std': [58.395, 57.12, 57.375], 'to_rgb': True}, {'type': 'GridMask', 'use_h': True, 'use_w': True, 'max_epoch': 20, 'rotate': 1, 'offset': False, 'ratio': 0.5, 'mode': 1, 'prob': 0.0, 'fixed_prob': True}, {'type': 'PointShuffle'}, {'type': 'DefaultFormatBundle3D', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'Collect3D', 'keys': ['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev'], 'meta_keys': ['camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix']}], 'test_pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.48, 0.48], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [0.0, 0.0], 'rand_flip': False, 'is_train': False}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [1.0, 1.0], 'rot_lim': [0.0, 0.0], 'trans_lim': 0.0, 'is_train': False}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'data/nuscenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ImageNormalize', 'mean': [103.53, 116.28, 123.675], 'std': [58.395, 57.12, 57.375], 'to_rgb': True}, {'type': 'DefaultFormatBundle3D', 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']}, {'type': 'Collect3D', 'keys': ['img', 'points', 'gt_bboxes_3d', 'gt_labels_3d', 'gt_masks_bev'], 'meta_keys': ['camera_intrinsics', 'camera2ego', 'lidar2ego', 'lidar2camera', 'lidar2image', 'img_aug_matrix', 'lidar_aug_matrix']}], 'data': {'samples_per_gpu': 4, 'workers_per_gpu': 4, 'train': {'type': 'CBGSDataset', 'dataset': {'type': 'NuScenesDataset', 'dataset_root': 'data/nuscenes/', 'ann_file': 'data/nuscenes/nuscenes_infos_train.pkl', 'pipeline': [{'type': 'LoadMultiViewImageFromFiles', 'to_float32': True}, {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'load_augmented': None}, {'type': 'LoadPointsFromMultiSweeps', 'sweeps_num': 9, 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32, 'pad_empty_sweeps': True, 'remove_close': True, 'load_augmented': None}, {'type': 'LoadAnnotations3D', 'with_bbox_3d': True, 'with_label_3d': True, 'with_attr_label': False}, {'type': 'ObjectPaste', 'stop_epoch': -1, 'db_sampler': {'dataset_root': 'data/nuscenes/', 'info_path': 'data/nuscenes/nuscenes_dbinfos_train.pkl', 'rate': 1.0, 'prepare': {'filter_by_difficulty': [-1], 'filter_by_min_points': {'car': 5, 'truck': 5, 'bus': 5, 'trailer': 5, 'construction_vehicle': 5, 'traffic_cone': 5, 'barrier': 5, 'motorcycle': 5, 'bicycle': 5, 'pedestrian': 5}}, 'classes': ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone'], 'sample_groups': {'car': 2, 'truck': 3, 'construction_vehicle': 7, 'bus': 4, 'trailer': 6, 'barrier': 2, 'motorcycle': 6, 'bicycle': 6, 'pedestrian': 2, 'traffic_cone': 2}, 'points_loader': {'type': 'LoadPointsFromFile', 'coord_type': 'LIDAR', 'load_dim': 5, 'use_dim': 5, 'reduce_beams': 32}}}, {'type': 'ImageAug3D', 'final_dim': [256, 704], 'resize_lim': [0.48, 0.48], 'bot_pct_lim': [0.0, 0.0], 'rot_lim': [0.0, 0.0], 'rand_flip': False, 'is_train': True}, {'type': 'GlobalRotScaleTrans', 'resize_lim': [1.0, 1.0], 'rot_lim': [0.0, 0.0], 'trans_lim': 0.0, 'is_train': True}, {'type': 'LoadBEVSegmentation', 'dataset_root': 'data/nuscenes/', 'xbound': [-50.0, 50.0, 0.5], 'ybound': [-50.0, 50.0, 0.5], 'classes': ['drivable_area', 'ped_crossing', 'walkway', 'stop_line', 'carpark_area', 'divider']}, {'type': 'RandomFlip3D'}, {'type': 'PointsRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectRangeFilter', 'point_cloud_range': [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]}, {'type': 'ObjectNameFilter', 'classes': ['car', 'truck', 'construction_veh [... cut ...]
--------------------------------------------------------------------------
--------------------------------------------------------------------------
mpirun detected that one or more processes exited with non-zero status, thus causing
the job to be terminated. The first process to do so was:

  Process name: [[33106,1],0]
  Exit code:    1
--------------------------------------------------------------------------
'
I don't know why it occurred and I wonder how to solve this bug.
Thank you very much!

kentang-mit, 2022-07-20:
First, it seems that you are using `v1.0-mini`. This is currently not quite recommended since I have only tested on `v1.0-trainval`. There are some people who succeeded in running evaluation on `v1.0-mini`, but you might have to search in the issues. 

Besides that, I would suggest you to start from simple models (such as the camera-only ones) and see whether the same error still occurs.

cherry-yh, 2022-07-21:
I just did evaluation using v1.0-mini on pure camera Det and Seg tasks, and succeeded. But i have other questions:
1)  when i use 'torchpack dist-run -np 8 python tools/test.py ', the process got stuck at: 
[image]
the code can only run succeed by setting -np=1, other np numbers have the same problem.
2) your map segmentation demo is great, which command can do this visualization? 
Thanks a lot.

cherry-yh, 2022-07-21:
@kentang-mit 
Thank you for your quick reply. 
For Q1, i have not figure out the problem temporarily; 
for Q2, I referred to issue #59 to visualize map segmentation result, and it really works! I appreciate your help very much. 

Sorry for some irrelevant issue: 
[https://github.com/mit-han-lab/bevfusion/blob/008b4a8b3287a6aba159712c4baff637e73d20e1/mmdet3d/datasets/pipelines/loading.py#L276](url)
I guess, during the generation of seg GT, some map elements such as "road_segment", "lane" all belongs to "drivable_area", and the binary labels need to be merged to one heatmap, but what does '*' mean in:
[image]

zhijian-liu, 2022-07-21:
Hi @cherry-yh, `drivable_area*` refers to the definition of drivable area used in previous papers, such as LSS and CVT. `drivable_area` follows the official definition as in NuScenes.

cherry-yh, 2022-07-21:
Ok, thanks @zhijian-liu , I will check the former papers for details, and dig deeper into the relationship between 'drivable_area*', 'drivable_area' representations and code realization.

kentang-mit, 2022-07-26:
Temporarily closed due to inactivity.

closed 2022-07-26 by kentang-mit