mit-han-lab/bevfusion #80: importerror for data preparation in colab

RouDanCongJii, 2022-07-21:
hallo,
when i use the cmd 
`!python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes --version v1.0-mini`

Traceback (most recent call last):
  File "tools/create_data.py", line 3, in <module>
    from data_converter import nuscenes_converter as nuscenes_converter
  File "/content/bevfusion/tools/data_converter/nuscenes_converter.py", line 13, in <module>
    from mmdet3d.core.bbox.box_np_ops import points_cam2img
  File "/content/bevfusion/mmdet3d/core/__init__.py", line 1, in <module>
    from .anchor import *  # noqa: F401, F403
  File "/content/bevfusion/mmdet3d/core/anchor/__init__.py", line 1, in <module>
    from mmdet.core.anchor import build_prior_generator
  File "/usr/local/lib/python3.8/site-packages/mmdet/core/__init__.py", line 3, in <module>
    from .bbox import *  # noqa: F401, F403
  File "/usr/local/lib/python3.8/site-packages/mmdet/core/bbox/__init__.py", line 8, in <module>
    from .samplers import (BaseSampler, CombinedSampler,
  File "/usr/local/lib/python3.8/site-packages/mmdet/core/bbox/samplers/__init__.py", line 10, in <module>
    from .score_hlr_sampler import ScoreHLRSampler
  File "/usr/local/lib/python3.8/site-packages/mmdet/core/bbox/samplers/score_hlr_sampler.py", line 3, in <module>
    from mmcv.ops import nms_match
  File "/usr/local/lib/python3.8/site-packages/mmcv/ops/__init__.py", line 2, in <module>
    from .assign_score_withk import assign_score_withk
  File "/usr/local/lib/python3.8/site-packages/mmcv/ops/assign_score_withk.py", line 5, in <module>
    ext_module = ext_loader.load_ext(
  File "/usr/local/lib/python3.8/site-packages/mmcv/utils/ext_loader.py", line 13, in load_ext
    ext = importlib.import_module('mmcv.' + name)
  File "/usr/local/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libtorch_cuda_cu.so: cannot open shared object file: No such file or directory

the following cmd are used to  install the environment
!conda create -n bevfusion python=3.8 -y
!activate bevfusion
!conda install pytorch==1.9.0 torchvision==0.10.0 torchaudio==0.9.0 -c pytorch
!pip install Pillow==8.4.0
!pip install tqdm
!pip install --upgrade --upgrade-strategy=only-if-needed torchpack
!pip install mmcv-full==1.4.0 -f https://download.openmmlab.com/mmcv/dist/cu111/torch1.9.0/index.html
!pip install mmdet==2.20.0
!pip install nuscenes-devkit
! git clone https://github.com/mit-han-lab/bevfusion.git
%cd /content/bevfusion
!python setup.py develop
!wget https://www.nuscenes.org/data/v1.0-mini.tgz
!tar -xf v1.0-mini.tgz -C /content/bevfusion/data/nuscenes

Is there anything wrong for the installation of the environment?

thanks

kentang-mit, 2022-07-21:
Hi @RouDanCongJii,

Thank you for your interest. I haven't tried setting up the environment on Colab but I will try it hopefully before the weekend. By the way, have you tried out the same process on a local machine? 

Best,
Haotian

RouDanCongJii, 2022-07-21:
hi,@kentang-mit 
I tried it and it worked. But it is installed on windows system.

kentang-mit, 2022-07-26:
I think it requires some effort to make the current codebase to work directly on Colab. We will probably wrap up the entire environment within a wheel and try again in the future. 

We suggest to perform evaluation locally.

closed 2022-07-26 by kentang-mit