mit-han-lab/bevfusion #82: N > 0 assert failed. CUDA kernel launch blocks must be positive, but got N= 0

tuan97ta, 2022-07-23:
I run test.py and get error : 
```
RuntimeError: /tmp/mmcv/mmcv/ops/csrc/common/utils/spconv/tensorview/helper_launch.h 14
N > 0 assert failed. CUDA kernel launch blocks must be positive, but got N= 0
```

kentang-mit, 2022-07-23:
Would you mind having a look at [this issue](https://github.com/mit-han-lab/bevfusion/issues/28)?

zzj403, 2022-07-29:
I've looked the mentioned issue #28 but I still can not fix the same problem.

cudatoolkit               11.3.1              h9edb442_10    conda-forge

numpy                     1.22.3           py38he7a7128_0    defaults
numpy-base                1.22.3           py38hf524024_0    defaults

numba                     0.55.2                   pypi_0    pypi
numpy                     1.22.3           py38he7a7128_0    defaults
numpy-base                1.22.3           py38hf524024_0    defaults
nuscenes-devkit           1.1.9                    pypi_0    pypi

mmcv-full                 1.4.0                     dev_0    <develop>
mmdet                     2.20.0                    dev_0    <develop>
mmdet3d                   0.0.0                     dev_0    <develop>

torchpack                 0.3.1                    pypi_0    pypi
torchvision               0.11.2               py38_cu113    pytorch
tornado                   6.2                      pypi_0    pypi

python_abi                3.8                      2_cp38    conda-forge
pytorch                   1.10.1          py3.8_cuda11.3_cudnn8.2.0_0    pytorch
pytorch-mutex             1.0                        cuda    pytorch

Some times this error comes out, but sometimes it doesn't come out and just leave my training stucked, and the utilization of GPU are (GPU0 100% GPU1 0% GPU2 100% ...GPU7 100% )
Sometimes stuck at around  20th step, sometimes stuck in later epoch .
I've met this problem many times. Sad...

kentang-mit, 2022-07-29:
I will start answering training-related questions after we release all the training code.

closed 2022-07-29 by kentang-mit

evgeniishch, 2022-09-28:
Hello! I encounter the same problem when running the training pipeline with
`torchpack dist-run -np 8 python tools/train.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml `

The official spconv repo says that [this error is caused by invalid conv params](https://github.com/traveller59/spconv/blob/8b52b3a9a6d78fc6168d688f9960d0329597c185/docs/COMMON_PROBLEMS.md). I would appreciate any ideas how this might be fixed, thank you!

cuge1995, 2022-10-24:
I solve it by `pip install spconv-cu102`

holestine, 2022-12-12:
I'm using a different version of CUDA so this worked for me pip install spconv-cu113

xiaotiancai899, 2023-06-10:
> pip install spconv-cu113

I have already installed the spconv-cu113, but the same error still exist. Any ideas about that? Thanks so much in advance！