mit-han-lab/bevfusion #83: What is the minimum memory  required for evaluation or training

XiangboGaoBarry, 2022-07-23:
I am using one rtxa4000 (16G) for evaluation,  
`torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox`
which seems ending up with "out of memory".
`mmdet3d/ops/spconv/src/indice_cuda.cu 124`
`cuda execution failed with error 2`

May I know what is minimum memory  required?

BaiLiping, 2022-07-23:
Same issue

BaiLiping, 2022-07-23:
the same error for lidar only detection. Can you just publish the detection file for NuScenes validate set and the test set?

XiangboGaoBarry, 2022-07-23:
BTW, I am able to run `tools/train.py` but does not have enough memory for `tools/test.py`. Seems weird to me.

kentang-mit, 2022-07-23:
Please take a look at [this issue](https://github.com/mit-han-lab/bevfusion/issues/26) for the OOM issue. It should be possible to run evaluation with 16GB GPU memory.

xxp912, 2022-07-25:
> BTW, I am able to run `tools/train.py` but does not have enough memory for `tools/test.py`. Seems weird to me.

Hello, do you train a model successfully by running `tools/train.py`?

VeeranjaneyuluToka, 2022-07-26:
I was able to run the inference using the below command
**torchpack dist+run -np 1 python tools/test.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml tools/pretrained/camera-only-det.pth --eval bbox**
but when i try to train using the below command by providing optimizer_config
**torchpack dist-run -np 1 python tools/train.py configs/nuscenes/det/centerhead/lssfpn/camera/256x704/swint/default.yaml**
It gives the following error
 **File "/home/hykeserver/Downloads/3d_OD/bevfusion-main/mmdet3d/models/vtransforms/lss.py", line 69, in get_cam_feats
    x = depth.unsqueeze(1) * x[:, self.D : (self.D + self.C)].unsqueeze(2)
RuntimeError: CUDA out of memory. Tried to allocate 610.00 MiB (GPU 0; 7.79 GiB total capacity; 3.13 GiB already allocated; 215.12 MiB free; 3.32 GiB reserved in total by PyTorch)
--------------------------------------------------------------------------
Primary job  terminated normally, but 1 process returned
a non-zero exit code. Per user-direction, the job has been aborted.
--------------------------------------------------------------------------**
I am using Quadro RTX 4000 gpu card with 8 GB. I think this can be solved when we reduced samples_per_gpu from 4 to 1 but when i try to reduce it is not getting changed. Is my thinking correct (changing samples_per_gpu wil resolve the above error). If it so, where should i change it to reflect in saved config file.

xxp912, 2022-07-26:
> 
Thank you for your reply, but I meet a mistake when I try to use the evaluation command provided by the readme.
And my signal GPU is RTX 3090.
```
~/bevfusion$ torchpack dist-run -np 1 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
ssh: Could not resolve hostname localhost:1: Temporary failure in name resolution
--------------------------------------------------------------------------
ORTE was unable to reliably start one or more daemons.
This usually is caused by:

* not finding the required libraries and/or binaries on
  one or more nodes. Please check your PATH and LD_LIBRARY_PATH
  settings, or configure OMPI with --enable-orterun-prefix-by-default

* lack of authority to execute on one or more specified nodes.
  Please verify your allocation and authorities.

* the inability to write startup files into /tmp (--tmpdir/orte_tmpdir_base).
  Please check with your sys admin to determine the correct location to use.

*  compilation of the orted with dynamic libraries when static are required
  (e.g., on Cray). Please check your configure cmd line and consider using
  one of the contrib/platform definitions for your system type.

* an inability to create a connection back to mpirun due to a
  lack of common network interfaces and/or no route found between
  them. Please check network connectivity (including firewalls
  and network routing requirements).
--------------------------------------------------------------------------
```
 @kentang-mit  Have you met this error?

VeeranjaneyuluToka, 2022-07-26:
No, i did not get this error. However this looks like something to do with SSH setup not with the BEVFusion source issue.

VeeranjaneyuluToka, 2022-07-26:
Wondering what is the minimum GPU memory required to train a model?

kentang-mit, 2022-07-26:
Hello @VeeranjaneyuluToka,

Training details will be released in the future. I currently do not have enough bandwidth to respond to training errors encountered using customized schedules. By the way, the minimum GPU memory required to train the fusion model is around 35GB, but we are working on reducing this requirement to less than 24GB. 

Best,
Haotian

kentang-mit, 2022-07-26:
Hello @xxp912,

Did you set up the OpenMPI framework on your machine?

Best,
Haotian

xxp912, 2022-07-27:
> 

Thank you very much! I solved the bug by install openmpi == 4.1.1 and mpi4py == 3.0.3. Besides, I suggest you could add the requirements to README.md. I believe that many people like me who have not used distributed training will face this problem.

xxp912, 2022-07-27:
> No, i did not get this error. However this looks like something to do with SSH setup not with the BEVFusion source issue.

Thank you for your suggesstions!

kentang-mit, 2022-07-27:
> @kentang-mit
> Thank you very much! I solved the bug by install openmpi == 4.1.1 and mpi4py == 3.0.3. Besides, I suggest you could add the requirements to README.md. I believe that many people like me who have not used distributed training will face this problem.

That's a very good point. Thanks for your suggestion and I will do it immediately.

closed 2022-07-27 by kentang-mit

Son-Goku-gpu, 2022-09-01:
> Hello @VeeranjaneyuluToka,
>
> Training details will be released in the future. I currently do not have enough bandwidth to respond to training errors encountered using customized schedules. By the [...]

Hi @kentang-mit , do you mean 35/24G for batch=4 or ??

flstahl, 2024-01-17:
> Hello @VeeranjaneyuluToka,
>
> Training details will be released in the future. I currently do not have enough bandwidth to respond to training errors encountered using customized schedules. By the [...]

Hi @kentang-mit ,

I am currently working with BEVFusion and having serious trouble with the GPU memory. I am having trouble understanding why this model needs so much memory for training. Given that the model size is ~150M params and let's assume an AdamW optimizer. 

I found that you would need:

- 4 bytes * number of parameters for fp32 training 
- 8 bytes * number of parameters for normal AdamW (maintains 2 states)

This comes to just shy of 2GB for the Model. Let's say we have a sample data size of 100 MB, I am having trouble understanding how we end up maxing out 24GB GPUs here. 

Is there something major I am missing?

Any help would be highly appreciated in helping me getting my head around this! Thanks a lot (@kapina)