mit-han-lab/bevfusion #86: mini result with have mAP of 0.5846, camera+lidar

BaiLiping, 2022-07-24:
Results writes to /data/home/scv1846/run/results_nusc.json
mAP: 0.5846
mATE: 0.4060
mASE: 0.4470
mAOE: 0.4739
mAVE: 0.4353
mAAE: 0.3202
NDS: 0.5841
Eval time: 2.8s

Per-class results:
Object Class    AP      ATE     ASE     AOE     AVE     AAE
car     0.926   0.171   0.158   0.096   0.115   0.073
truck   0.810   0.148   0.129   0.049   0.060   0.000
bus     0.994   0.187   0.078   0.023   0.514   0.361
trailer 0.000   1.000   1.000   1.000   1.000   1.000
construction_vehicle    0.000   1.000   1.000   1.000   1.000   1.000
pedestrian      0.933   0.120   0.256   0.323   0.206   0.127
motorcycle      0.722   0.190   0.260   0.393   0.052   0.000
bicycle 0.569   0.163   0.241   0.382   0.535   0.000
traffic_cone    0.892   0.081   0.348   nan     nan     nan
barrier 0.000   1.000   1.000   1.000   nan     nan
{'object/car_ap_dist_0.5': 0.8429, 'object/car_ap_dist_1.0': 0.9411, 'object/car_ap_dist_2.0': 0.9541, 'object/car_ap_dist_4.0': 0.9649, 'object/car_trans_err': 0.1713, 'object/car_scale_err': 0.1579, 'object/car_orient_err': 0.0957, 'object/car_vel_err': 0.1155, 'object/car_attr_err': 0.0733, 'object/mATE': 0.406, 'object/mASE': 0.447, 'object/mAOE': 0.4739, 'object/mAVE': 0.4353, 'object/mAAE': 0.3202, 'object/truck_ap_dist_0.5': 0.766, 'object/truck_ap_dist_1.0': 0.8072, 'object/truck_ap_dist_2.0': 0.8276, 'object/truck_ap_dist_4.0': 0.8383, 'object/truck_trans_err': 0.1479, 'object/truck_scale_err': 0.129, 'object/truck_orient_err': 0.0486, 'object/truck_vel_err': 0.0603, 'object/truck_attr_err': 0.0, 'object/construction_vehicle_ap_dist_0.5': 0.0, 'object/construction_vehicle_ap_dist_1.0': 0.0, 'object/construction_vehicle_ap_dist_2.0': 0.0, 'object/construction_vehicle_ap_dist_4.0': 0.0, 'object/construction_vehicle_trans_err': 1.0, 'object/construction_vehicle_scale_err': 1.0, 'object/construction_vehicle_orient_err': 1.0, 'object/construction_vehicle_vel_err': 1.0, 'object/construction_vehicle_attr_err': 1.0, 'object/bus_ap_dist_0.5': 0.9943, 'object/bus_ap_dist_1.0': 0.9943, 'object/bus_ap_dist_2.0': 0.9943, 'object/bus_ap_dist_4.0': 0.9943, 'object/bus_trans_err': 0.187, 'object/bus_scale_err': 0.0775, 'object/bus_orient_err': 0.0232, 'object/bus_vel_err': 0.5142, 'object/bus_attr_err': 0.3615, 'object/trailer_ap_dist_0.5': 0.0, 'object/trailer_ap_dist_1.0': 0.0, 'object/trailer_ap_dist_2.0': 0.0, 'object/trailer_ap_dist_4.0': 0.0, 'object/trailer_trans_err': 1.0, 'object/trailer_scale_err': 1.0, 'object/trailer_orient_err': 1.0, 'object/trailer_vel_err': 1.0, 'object/trailer_attr_err': 1.0, 'object/barrier_ap_dist_0.5': 0.0, 'object/barrier_ap_dist_1.0': 0.0, 'object/barrier_ap_dist_2.0': 0.0, 'object/barrier_ap_dist_4.0': 0.0, 'object/barrier_trans_err': 1.0, 'object/barrier_scale_err': 1.0, 'object/barrier_orient_err': 1.0, 'object/barrier_vel_err': nan, 'object/barrier_attr_err': nan, 'object/motorcycle_ap_dist_0.5': 0.6608, 'object/motorcycle_ap_dist_1.0': 0.7164, 'object/motorcycle_ap_dist_2.0': 0.742, 'object/motorcycle_ap_dist_4.0': 0.7689, 'object/motorcycle_trans_err': 0.19, 'object/motorcycle_scale_err': 0.2603, 'object/motorcycle_orient_err': 0.3928, 'object/motorcycle_vel_err': 0.0516, 'object/motorcycle_attr_err': 0.0, 'object/bicycle_ap_dist_0.5': 0.559, 'object/bicycle_ap_dist_1.0': 0.559, 'object/bicycle_ap_dist_2.0': 0.559, 'object/bicycle_ap_dist_4.0': 0.6007, 'object/bicycle_trans_err': 0.1628, 'object/bicycle_scale_err': 0.241, 'object/bicycle_orient_err': 0.3817, 'object/bicycle_vel_err': 0.535, 'object/bicycle_attr_err': 0.0, 'object/pedestrian_ap_dist_0.5': 0.9171, 'object/pedestrian_ap_dist_1.0': 0.9262, 'object/pedestrian_ap_dist_2.0': 0.9344, 'object/pedestrian_ap_dist_4.0': 0.9535, 'object/pedestrian_trans_err': 0.1199, 'object/pedestrian_scale_err': 0.2556, 'object/pedestrian_orient_err': 0.3234, 'object/pedestrian_vel_err': 0.2056, 'object/pedestrian_attr_err': 0.1266, 'object/traffic_cone_ap_dist_0.5': 0.8918, 'object/traffic_cone_ap_dist_1.0': 0.8918, 'object/traffic_cone_ap_dist_2.0': 0.8918, 'object/traffic_cone_ap_dist_4.0': 0.8918, 'object/traffic_cone_trans_err': 0.0807, 'object/traffic_cone_scale_err': 0.3483, 'object/traffic_cone_orient_err': nan, 'object/traffic_cone_vel_err': nan, 'object/traffic_cone_attr_err': nan, 'object/nds': 0.5840639091317819, 'object/map': 0.5845882409495385}
(openmmlab) [scv1846@ln01 tools]$ connection error

BaiLiping, 2022-07-24:
I tried the test.py file with mini dataset, with the camera+lidar detector. The result seems to be worse than reported? just to check of all the specifics are correct?

kentang-mit, 2022-07-24:
All results in our paper are reported on `v1.0-trainval`.

closed 2022-07-24 by kentang-mit