mit-han-lab/bevfusion #90: About point depth used in DepthLSSTransform

chenchaoxu, 2022-08-05:
Hi talents,

I have a question about DepthLSSTransform used in your bevFusion model. It seems that info of point depth is used for LSS. Dose it get large gain of performance comparing to the orginal LSS? Plus, why not chosing to set point depth as surpervision instead?

kentang-mit, 2022-08-05:
Hello @chenchaoxu,

Thanks for your interest. We did not notice a large gain of performance comparing to the original LSS and this is the reason why we did not talk about it in the paper. Your suggestion is actually a very good idea, we definitely want to try that in the future. We have tried directly converting the 2.5D depth to a one-hot representation (so that you do not have to learn it and can directly use the "GT"), but I remember that did not work out. 

Best,
Haotian

kentang-mit, 2022-08-08:
Temporarily closed due to inactivity.

closed 2022-08-08 by kentang-mit

hehern, 2024-02-02:
There is a bug in  [https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/models/vtransforms/base.py](url). L284, L290 will change the origin lidar points.