mit-han-lab/bevfusion #91: There is a strange question occurring

qfwysw, 2022-08-06:
[image]
My code:
`torchpack dist-run -np 8 python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
`
Looking at previous questions, I have openmpi installed.

kentang-mit, 2022-08-06:
Hello @qfwysw,

Would you mind running the following command again and provide the error message?

```bash
torchpack dist-run -np 8 -v python tools/test.py configs/nuscenes/det/transfusion/secfpn/camera+lidar/swint_v0p075/convfuser.yaml pretrained/bevfusion-det.pth --eval bbox
```

Best,
Haotian

qfwysw, 2022-08-06:
[image]
Just like this.
But, I dont run another program.

qfwysw, 2022-08-06:
[image]
I try to do this, but it doesn't work.

kentang-mit, 2022-08-06:
Probably you can try launching the script when there is no other distributed job running.

qfwysw, 2022-08-06:
No matter which port I choose, he prompts me that the address is already in use

qfwysw, 2022-08-06:
[image]
[image]
I have chose another port, it is really strange.

kentang-mit, 2022-08-06:
Probably changing the `MASTER_HOST` will not work, I remember that @zhijian-liu mentioned that to me a while ago.

kentang-mit, 2022-08-08:
Temporarily closed due to inactivity. Feel free to reopen if new issues occur.

closed 2022-08-08 by kentang-mit