mit-han-lab/bevfusion #92: no singapore-onenorth.json

qfwysw, 2022-08-06:
[image]
I check my dataset, I didn't fint it. What's wrong with this?

kentang-mit, 2022-08-06:
You will have to download the full map extension from nuScenes.

closed 2022-08-06 by kentang-mit

qfwysw, 2022-08-10:
[image]
[image]
Except full dataset，do I have to download the Map expansion? Just like the picture.

youchaoqin, 2023-02-09:
Which version of the map expansion should we use? Is v1.3 ok?