mit-han-lab/bevfusion #95: Address alread used

qfwysw, 2022-08-10:
If I set the gpu 1, the program can run. However, when I set the num > 1, the code will notes me address already used.
I print the MASTER_HOST, every thread uses the same MASTER_HOST.
So what can I do ?

kentang-mit, 2022-08-10:
I just want to confirm whether there are other distributed jobs running on your machine, as is mentioned in the previous issue.

qfwysw, 2022-08-10:
Sorry, I am not sure if there are other distributed programs running, I am running this program on a cluster.

kentang-mit, 2022-08-10:
You may just use `nvidia-smi` to check out whether there are other jobs running on the machine. It will be better if you can test whether you will still meet with this error when the GPUs are vacant.

qfwysw, 2022-08-10:
I check it just now, no other program.
[image]
[image]

qfwysw, 2022-08-10:
[image]
I don't know if output like this means my mpirun installation was successful.

zhijian-liu, 2022-08-10:
You could add `-H localhost:<p>` with different `<p>` to avoid port conflicts.

qfwysw, 2022-08-11:
Thanks for your help!

qfwysw, 2022-08-12:
[image]
[image]
I set the H, but I get the same error.
If I set the gpu 1, the code works fine.

qfwysw, 2022-08-12:
I slove the problem. I replaced train.py in the code with train.py of mmdetection3d

closed 2022-08-12 by qfwysw

zhenghw111, 2022-12-13:
> 

Hi ！I have encountered the same problem. How do you replace the train.py file? Need to change somewhere? Look forward to your reply

zhenghw111, 2022-12-13:
> I slove the problem. I replaced train.py in the code with train.py of mmdetection3d

Hello, I have encountered the same problem. How do you replace the train.py file? Need to change somewhere? Look forward to your reply

huzhihen, 2023-12-15:
> I slove the problem. I replaced train.py in the code with train.py of mmdetection3d

Hello, I have encountered the same problem. Can you tell me how to modify the train.py? Thank you very much!

Sranc3, 2024-01-03:
> > I slove the problem. I replaced train.py in the code with train.py of mmdetection3d
> 
> Hello, I have encountered the same problem. Can you tell me how to modify the train.py? Thank you very much!

Any suggestions? I met the same problem on  two of my machines, I can only use one GPU for training and testing.

REFRAING, 2024-04-13:
> I slove the problem. I replaced train.py in the code with train.py of mmdetection3d
Hello, I meet same problem, could you tell me how do you replace the train.py？ Thanks