mit-han-lab/bevfusion #96: Some questions about ObjectSample

miaochunle, 2022-08-10:
I visualized the pasted object and found that the paste object was wrong. This problem has been bothering me for a long time. I really hope to get your help

Objectsample GT visualization results are as follows: 
[image]

The following is the code used:

https://github.com/mit-han-lab/bevfusion/blob/main/mmdet3d/datasets/pipelines/dbsampler.py

 -
    type: ObjectPaste
    stop_epoch: ${gt_paste_stop_epoch}
    db_sampler:
      dataset_root: ${dataset_root}
      info_path: ${dataset_root + "nuscenes_dbinfos_train.pkl"}
      rate: 1.0
      prepare: 
        filter_by_difficulty: [-1]
        filter_by_min_points:
          car: 5
          truck: 5
          bus: 5
          trailer: 5
          construction_vehicle: 5
          traffic_cone: 5
          barrier: 5
          motorcycle: 5
          bicycle: 5
          pedestrian: 5
      classes: ${object_classes}
      sample_groups:
        car: 2
        truck: 3
        construction_vehicle: 7
        bus: 4
        trailer: 6
        barrier: 2
        motorcycle: 6
        bicycle: 6
        pedestrian: 2
        traffic_cone: 2
      points_loader:
        type: LoadPointsFromFile
        coord_type: LIDAR
        load_dim: ${load_dim}
        use_dim: ${use_dim}
        reduce_beams: ${reduce_beams}

kentang-mit, 2022-08-10:
Hi @miaochunle,

That is quite interesting. The current object paste code is exactly what we used in our experiments, and we are able to reproduce some results from previous papers. Would you mind kindly share with me the visualization code you are using? One thing you probably need to note is that the coordinate system used in this codebase is different from mmdetection3d.

Best,
Haotian

miaochunle, 2022-08-11:
> 
In ObjectPaste.__call__, I added two codes to save separately data_gt and paste_ gt. Locally, visualize with the same script. I found data_ GT is correct, paste_ GT is wrong, I feel very confused. I have also considered whether it is a coordinate system, but the same visualization script should not only have paste_ GT is wrong. I feel very helpless

        if self.stop_epoch is not None and self.epoch >= self.stop_epoch:
             return data
        vision_flag = True
        gt_bboxes_3d = data["gt_bboxes_3d"]
        gt_labels_3d = data["gt_labels_3d"]

        if vision_flag:
            import copy
            gt_bboxes_3d_ = copy.deepcopy(gt_bboxes_3d.tensor)
            points_ = copy.deepcopy(data["points"])
            gt_bboxes_3d_[..., 2] -= gt_bboxes_3d_[..., 5] / 2
            gt_bboxes_3d_ = LiDARInstance3DBoxes(gt_bboxes_3d_, box_dim=9)
            bboxes = gt_bboxes_3d_.corners.numpy()
            save_name = "cutandpaste/" + (data["lidar_path"].split("/")[-1]).split(".")[0] + "_bboxes_gt"
            np.save(save_name, bboxes)
            save_name = "cutandpaste/" + (data["lidar_path"].split("/")[-1]).split(".")[0] + "_point_gt"
            np.save(save_name, points_.coord.numpy())

        if sampled_dict is not None:
            sampled_gt_bboxes_3d = sampled_dict["gt_bboxes_3d"]
            sampled_points = sampled_dict["points"]
            sampled_gt_labels = sampled_dict["gt_labels_3d"]
            print()

            if vision_flag:
                import copy
                sampled_gt_bboxes_3d_ = copy.deepcopy(sampled_gt_bboxes_3d)
                sampled_points_ = copy.deepcopy(sampled_points)
                sampled_gt_bboxes_3d_[..., 2] -= sampled_gt_bboxes_3d_[..., 5] / 2
                sampled_gt_bboxes_3d_ = LiDARInstance3DBoxes(sampled_gt_bboxes_3d_, box_dim=9)
                bboxes = sampled_gt_bboxes_3d_.corners.numpy()
                save_name = "cutandpaste/" + (data["lidar_path"].split("/")[-1]).split(".")[0] + "_bboxes_paste_before"
                np.save(save_name, bboxes)
                save_name = "cutandpaste/" + (data["lidar_path"].split("/")[-1]).split(".")[0] + "_point_paste_before"
                np.save(save_name, sampled_points_.coord.numpy())

The following are the visualization results:

paste_gt:
[image]

data_gt:
[image]

The visualization code is as follows:

    pcds = np.load(
        r"1105\n008-2018-08-28-16-16-48-0400__LIDAR_TOP__1535488135946939_point_gt.npy")

    gt_box3d_corners_aug = np.load(
        r"1105\n008-2018-08-28-16-16-48-0400__LIDAR_TOP__1535488135946939_bboxes_gt.npy")
    vis_3d(pcds, gt_box3d_corners=gt_box3d_corners_aug)

    def vis_3d(pcds, pcds_color=None, gt_box3d_corners=None, pred_box3d_corners=None, pc_radar=None, x_range=(-80, 80), y_range=(-80, 80), z_range=(-3, 5)):
    # pcds numpy float32 array (n, c)
    # pcds_color uint8 array (n, 3) denotes the color of each point 0~255
    FOR1 = o3d.geometry.TriangleMesh.create_coordinate_frame(size=5, origin=[
                                                             0, 0, 0])

    if (pc_radar is not None) and (len(pc_radar) > 0):
        pc_tmp = np.pad(pcds[:, :3], ((0, len(pc_radar)),
                        (0, 0)), 'constant', constant_values=0)
        pc_tmp[-len(pc_radar):, :2] = pc_radar[:, :2]

        pcd_l = o3d.geometry.PointCloud()
        pcd_l.points = o3d.utility.Vector3dVector(pc_tmp)
        if pcds_color is not None:
[... 54 lines cut ...]
    if pc_radar is not None:
        for idx in range(len(pc_radar)):
            line_set = o3d.geometry.LineSet()
            p_tmp = np.zeros((2, 3), dtype=np.float32)
            p_tmp[0, :2] = pc_radar[idx, [0, 1]]
            p_tmp[1, :2] = pc_radar[idx, [0, 1]] + pc_radar[idx, [2, 3]]
            line_set.points = o3d.utility.Vector3dVector(p_tmp)
            line_set.lines = o3d.utility.Vector2iVector([[0, 1]])
            line_set.colors = o3d.utility.Vector3dVector([[0, 0, 0]])
            line_sets.append(line_set)

    line_set = o3d.geometry.LineSet()
    corners = []
    for z in z_range:
        corners.append([x_range[0], y_range[0], z])
        corners.append([x_range[0], y_range[1], z])
        corners.append([x_range[1], y_range[1], z])
        corners.append([x_range[1], y_range[0], z])

    line_set.points = o3d.utility.Vector3dVector(corners)
    colors = [[0, 0, 0] for i in range(len(lines))]
    line_set.lines = o3d.utility.Vector2iVector(lines)
    line_set.colors = o3d.utility.Vector3dVector(colors)
    line_sets.append(line_set)
    o3d.visualization.draw_geometries([*line_sets, FOR1, pcd_l])

miaochunle, 2022-08-17:
We have solved this problem. I'm very sorry, because we used the data conversion of other projects, the coordinate system of paste gt database was wrong.

kentang-mit, 2022-08-17:
I see, I'm glad that the problem is resolved. I will temporarily close this issue.

closed 2022-08-17 by kentang-mit