mit-han-lab/bevfusion #97: Some question about the dataset

qfwysw, 2022-08-11:
I'm so sorry to bother you. I have some difficulties.
I have download the map extention like below. Do I need to generate data according to your code? My original data generation is also before the coordinate system is changed.
[image]
[image]
[image]
I have no idea to slove this. Could you help me ?

kentang-mit, 2022-08-11:
Would you mind trying out the data preprocessing pipeline mentioned in the README? Please make sure that all the library versions match our descriptions in the README (especially mmcv and mmdet).

qfwysw, 2022-08-12:
I have done this, but the error is still here.  I'm not quite sure if the dataset I downloaded is correct. 
[image]
[image]

qfwysw, 2022-08-12:
Thanks for your help, I slove the problems.

closed 2022-08-12 by qfwysw