mit-han-lab/bevfusion #99: Reproduction of the lidar branch

Zhongdao, 2022-08-12:
Hi, nice work! 

Since the training details are not released yet, I'm here to share some of my progresses, and hopefully, discuss with you guys a bit.
I reproduced the lidar branch and get a number of around 63 mAP on nuscenes val, around 2 points lower than the reported number.
To get this number, I followed the exact [config](https://github.com/XuyangBai/TransFusion/blob/master/configs/transfusion_nusc_voxel_L.py) in TransFusion.  
A bit weird that there is still 2 points lower.

The only difference that I can see is in the "fading strategy" of GT sampling augmentation. In the implementation of TransFusion, the last 5 epoch training starts with lr=1e-4; while in contrast, in this implementation, the first 15 epoch and the last 5 shares a continuous LR schedule.

Anyone has any idea on this?

kentang-mit, 2022-08-12:
I think your schedule is fine. We did use a continuous LR schedule in our experiments. By the way, probably you can try running the test script separately after training.

Zhongdao, 2022-08-14:
That's definitely a good point. By separate evaluation, the results look a bit better, but still lower than the expectation. 
I cannot figure out the exact reason by far, so looking forward to the training details!

qfwysw, 2022-08-15:
> Hi, nice work!
>
> Since the training details are not released yet, I'm here to share some of my progresses, and hopefully, discuss with you guys a bit. I reproduced the lidar branch and get a number [...]

Hi, did you just add trasfution's lr_config in voxelnet_0p075.yaml? I just do like this in convfuser.yaml. However, there are some problems. Have you encountered these problems?
[image]

Zhongdao, 2022-08-15:
Hi @qfwysw 
Besides adding lr_config and momentum config, I find the most important hyparameters to be tuned are the ones for data augmentation. 
Regarding your problem, it seems the cost matrix contains non-numeric element. 
I cannot identify the real reason, but if you check the cost matrix carefully, it should be not difficult to find out.

Zhongdao, 2022-08-15:
Random scale, rotation, translation, and GT paste.
See Line 30 to Line 76 in https://github.com/XuyangBai/TransFusion/blob/master/configs/transfusion_nusc_voxel_L.py

qfwysw, 2022-08-15:
Thank you for your generous answer. This helped me a lot!

kentang-mit, 2022-08-15:
> Hi, did you just add trasfution's lr_config in voxelnet_0p075.yaml? I just do like this in convfuser.yaml. However, there are some problems. Have you encountered these problems? [image]

Just to clarify, our finetuning schedule is different from TransFusion. Please stay tuned, we are working on cleanning the training code.

qfwysw, 2022-08-16:
Thanks for the reminder, it should be the reason of lr config.

kentang-mit, 2022-08-25:
Temporarily closed due to inactivity.

closed 2022-08-25 by kentang-mit

IAMShashankk, 2022-09-03:
@Zhongdao Is it possible for you to share your config for training?