mit-han-lab/bevfusion #101: CUDA kernel errors

TheFatBlue, 2022-08-15:
Hello, I encountered the same problem as solved in [issue 26](https://github.com/mit-han-lab/bevfusion/issues/26) when I trained the model with the command "torchpack dist-run -np 8 python tools/train.py configs/nuscenes/seg/camera-bev256d2.yaml". I also tried to evaluate the pretrained model and the replacement mentioned in that issue, and all the performances are the same. However, after removing all `-gencode=arch=compute_XX,code=sm_XX` in the `setup.py` in my case, the error still exist.

Here is the error info:

File "PATH/bevfusion/mmdet3d/ops/bev_pool/bev_pool.py", line 96, in bev_pool
    x = x.permute(0, 4, 1, 2, 3).contiguous()
RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call,so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.

I used the M40, cuda version is 10.2, pytorch version is 1.10.0
Could you help me fix the problem? thanks

TheFatBlue, 2022-08-16:
I did so but the problem still exists

kentang-mit, 2022-08-16:
Did you rerun the setup develop command?

TheFatBlue, 2022-08-16:
I did... I'm re-downloading project and rerun the setup, hope it will work

kentang-mit, 2022-08-17:
Perhaps you can try another idea: for [this line](https://github.com/mit-han-lab/bevfusion/blob/c53e2283b0ebd00b5adbed8b3de0d39362ad3287/setup.py#L22), you may replace `compute_70` and `sm_70` with `compute_52` and `sm_52` (corresponding to M40). And you can remove lines 23-25.

TheFatBlue, 2022-08-17:
thank you very much and I solved the problem. Previously, I encountered the same problem as this [issue](https://github.com/mit-han-lab/bevfusion/issues/81), and I just replace the code [here](https://github.com/mit-han-lab/bevfusion/blob/main/configs/nuscenes/default.yaml#L261) with `  samples_per_gpu: 2`. This time, I also replace the next line with `  workers_per_gpu: 2`, and did the replacement you mentioned, and it works. I don't know exactly because of which method but it did work.

closed 2022-08-17 by TheFatBlue