mit-han-lab/bevfusion #102: how to download the map extension

junwenjiao, 2022-08-17:
when I try `python tools/create_data.py nuscenes --root-path ./data/nuscenes --out-dir ./data/nuscenes --extra-tag nuscenes`.I can not find nuscenes_infos_xxx_mono3d.coco.json.And I do not find how to download the map extension.
wish your answer.

qfwysw, 2022-08-17:
[image]
Hi, you can get it in nuscene.

junwenjiao, 2022-08-17:
just unzip it?but I can not find the right folder structure like  author said.

junwenjiao, 2022-08-17:
mmdetection3d
├── mmdet3d
├── tools
├── configs
├── data
│   ├── nuscenes
│   │   ├── maps
│   │   ├── samples
│   │   ├── sweeps
│   │   ├── v1.0-test
|   |   ├── v1.0-trainval
This one.

junwenjiao, 2022-08-17:
I can not find nuscenes_infos_xxx_mono3d.coco.json.And I see the create_data.py but find something about json.

kentang-mit, 2022-08-17:
The mono3d file is not needed for this project.

junwenjiao, 2022-08-17:
thanks for your help.Best wish!

kentang-mit, 2022-08-17:
Great, I will temporarily close this issue because it is resolved. Feel free to reopen if you have further questions.

closed 2022-08-17 by kentang-mit